riscvarchive/riscv-code-size-reduction #108: C.NOT instruction not part of the  Zcee-1.0-rc1 release candidate.

RogierBrussee, 2021-08-13:
I saw the you are preparing the Zcee for release. The C.NOT instruction does not seem to be part of it. Is this oversight or is there a reason?

jnk0le, 2021-08-13:
Probably due to not great results. Zbb also has logic instruction with negate on input so there is even less value in 'c.not'.
OTOH compilers are not aware of `c.not` and may not generate expected sequence to contribute c.not.

@tariqkurd-repo how is the analysis of those 1.13% of llvm picojpeg?

tariqkurd-repo, 2021-08-13:
Zcee won't be released seperately in the short term - we're going to wait for toolchain support and then assess the different instructions in the autumn.

If C.NOT is included then I agree that it should be in the simplest subset for high performance cores (Zcee).

As @jnk0le mentioned, the real question is whether we decide to include it (it is very cheap after all, and occasionally useful).

I'll close this, as the subsets are not finalised yet.

closed 2021-08-13 by tariqkurd-repo

RogierBrussee, 2021-08-13:
I saw in the benchmarks that it has a similar low absolute, but good bang for the encoding bit characteristic as the C.SEXT.B... C.ZEXT.H instructions, and  had a Y in the "is it worth it column" so I wondered. 

I must admit its attraction is partly aesthetics: it is a bit odd that C.AND, C.OR and C.XOR exist but C.NOT does odd. Not a very good technical reason, and ANDN, ORN and NXOR should avoid many of its uses (if the compiler is taught), but Zcee will require introducing the unitary C.OP rsd' encoding format anyway so that makes it much lower cost than in the original C spec. 

And just as an idea, the same goes for a hypothetical C.NEZ and C.EQZ instruction which obviously have to be benchmarked first.