riscvarchive/riscv-code-size-reduction #115: Usefulness of return value field of POP instruction

Lukacma, 2021-09-07:
Specification mentions that POP instruction has ret_val field, which can be used to set return value of the function. However, in presence of POPRET instruction, the only time when POP is beneficial to be used is when tail call is present. In such situations, the return value is supplied by tail call instead of the caller. Therefore, I do not think there exist situations when the ret_val field will  be used and maybe it would be good idea to use it for something else.

tariqkurd-repo, 2021-09-27:
we should remove the ret_val field from the POP encodings.

tariqkurd-repo, 2021-10-04:
This version of the spec has this fixed

https://github.com/riscv/riscv-code-size-reduction/blob/after_v0.50.1_dev/ISA%20proposals/Huawei/Zce_spec.adoc

closed 2021-10-04 by tariqkurd-repo