riscvarchive/riscv-code-size-reduction #116: C.PUSH/PUSH in LLVM doesn't support areg_list

tariqkurd-repo, 2021-09-27:
The LLVM implementation of C.PUSH/PUSH doesn't support `areg_list`, which is this part of the encoding:

c.push   {ra,s0-s1}, `{a0-a1},`    -96; //store ra,s0-s1;  `mv s0,a0; mv s1,a1; `  decrement sp by 96

They have been excluded because they restrict the move instructions to moving to specifically numbered `s` registers.
C.PUSH/PUSH are inferred after the register allocation, and no attempt is made to change which `s` registers the `a` registers are moved into.

The definition of C.PUSH is inflexible to save encoding space - `areg_list` is automatically inferred from `reg_list` (`{ra,s0-s1}` in this example).

so we need to know if it's possible to allocate the target `s` registers to allow `areg_list` to be included. If it's *not* possible:

1. should we remove `areg_list` from the syntax entirely (it saves > 1% of code size)?
2. should we allow `areg_list` to be optional in the 16-bit encoding? and only use it if it matches the registers which the register allocator chose (clearly costs encoding space).
3. should we invent an instruction a bit like `c.mva01s07` to implement `areg_list` in a more flexible way

tariqkurd-repo, 2021-09-27:
Compiler feedback is that we should drop areg_list, and find another solution.

jnk0le, 2021-09-27:
The compiler allocator can simply ignore that `push` instruction is moving anything to `s` registers (no codesize savings in this case).
OTOH, reg moves have to be uninterruptible sequence with a bit of added sequencer complexity.

Silabs-ArjanB, 2021-09-28:
> The LLVM implementation of C.PUSH/PUSH doesn't support areg_list, which is this part of the encoding:
>
> c.push {ra,s0-s1}, {a0-a1}, -96; //store ra,s0-s1; mv s0,a0; mv s1,a1; decrement sp by 96
>
> They [...]

Does the same hold for GCC?

Even if the compiler does not (yet) take register allocation into account, just keeping the c.push/push definition as it is should allow the compiler to achieve the same results as the Python script, right (or did the Python script make some assumption on how registers might have been allocated)? And then it can only get better in the future if the register alocator can hopefully be adapted.

Would the longer term plan for LLVM be to actually optimize register allocation for instructions as C.PUSH? Similarly, ideally the compiler performs register allocation knowing which registers are usable in regular C extension instructions.

If compiler experts think that LLVM _can_ in the future take instructions as C.PUSH/PUSH into account during its register allocation, then I would be very much in favor of keeping their current specification.

> so we need to know if it's possible to allocate the target s registers to allow areg_list to be included. If it's not possible:
> 
>     should we remove areg_list from the syntax entirely (it saves > 1% of code size)?
> 

That would be a big loss. I would keep areg_list included in the specification if we can reasonable assume that compilers can (possibly at a later date) perform register allocation to benefit from it.

>     should we allow areg_list to be optional in the 16-bit encoding? and only use it if it matches the registers which the register allocator chose (clearly costs encoding space).

Making it optional costs even more space. The answer really depends on whether the compiler(s) will be eventually architected to benefit from such instructions (in which case I think we should not spend additional code space on making it optional, but just accept that in the shorter term the compiler does not fully utilize the potential of the instruction).

>>    should we invent an instruction a bit like c.mva01s07 to implement areg_list in a more flexible way

This will also reduce the code size savings.

tariqkurd-repo, 2021-09-28:
The feedback I'm given is that it's not feasible to implement in LLVM (with noindication that the situation will improve in the future). Unfortunately areg_list is not optional in the 16-bit encodings, so to make it optional would clearly cost more encoding space. Also GCC and LLVM are not the only compilers out there, the final spec must be implementable in all compilers.

abukharmeh, 2021-09-28:
(or did the Python script make some assumption on how registers might have been allocated ?)  ==> The Python script assumed very flexible registers allocation, because we were told then by somebody its possible to achieve (I found that strange assumption then thus I  wrote a note in the script noting that). When they actually started implementing this, they said it would be very hard to achieve such flexibility.

tariqkurd-repo, 2021-10-04:
I thin the sensible solution here is to remove areg_list from the PUSH specification and implement C.MVS07A01, i.e. move a0-a1 into s0-7. This solution fits well with the compiler, and doesn't have the srtange side-effect of overwriting s registers which were not actually needed. Closing for now, can reopen this if necessary.

https://github.com/riscv/riscv-code-size-reduction/blob/after_v0.50.1_dev/ISA%20proposals/Huawei/Zce_spec.adoc

this version of the spec has the updates

tariqkurd-repo, 2021-10-05:
@topperc do you have any comments on this?

topperc, 2021-10-05:
> @topperc do you have any comments on this?

I don't know how to implement this in LLVM's register allocation today. But I admit register allocation is not my strongest area. I wasn't sure from the discussion if gcc is able to do this? I can try talking to LLVM register allocation experts if you want.

kito-cheng, 2021-10-06:
> Are you saying that you would have the c.push copy a0 to s0 and a1 to s1 in hopes that you can erase a redundant mv later? Or you would calculate the areg_list based on the existence of the mvs?

Always emit c.push with copy a0 to s0 and a1 to s1 and optimize that later, GCC has a hard register copy propagation after RA and prologue/epilogue generation, I expect that could work on this, and those arg_list could be dropped in peephole/split* pass if `s0` and `s1` are unused.

* This is GCC implementation trick, but I think we can did some register liveness analysis to convert c.push w/ arg_list to c.push w/o arg_list in LLVM too.

topperc, 2021-10-06:
> Always emit c.push with copy a0 to s0 and a1 to s1 and optimize that later, GCC has a hard register copy propagation after RA and prologue/epilogue generation, I expect that could work on this, and [...]

We can probably do the same in LLVM, but I think the concern raised in this issue is that there is no guarantee that the exact `mv s0, a0` and `mv s1, a1` will be emitted. If `a0` and `a1` are moved to other s registers there is no code size savings. Can gcc make a0 prefer to copy to s0?

kito-cheng, 2021-10-06:
> If a0 and a1 are moved to other s registers there is no code size savings. Can gcc make a0 prefer to copy to s0?

Yeah, it's hard to guarantee the register allocator could always prefer `a0` copy to `s0`, a quick solution come to my mind is emit those move (s0 <- a0 and s1 <- a1) at early stage of back-end, and then let copy propagation to propagate that, some use of pseudo/virtual register will become hard register before RA, but this might not optimal way since that might interference RA's work.

tariqkurd-repo, 2021-10-06:
Hi @topperc , @kito-cheng , if we have this kind of situation:

```
#From debian / re compiled with GCC (RV64), reordered to put the registers in order
    53be:       7135                    addi    sp,sp,-160
    53c6:       ed06                    sd      ra,152(sp)
    53c8:       e922                    sd      s0,144(sp)
...s1-s10...
    53d8:       fc6e                    sd      s11,56(sp)
    53da:       84aa                    mv      s0,a0 <- will be covered by PUSH
    53e0:       892e                    mv      s2,a1 <- will not covered by PUSH
```

and we generate this code:

```
    53be:       <16-bit>                c.push  {ra,s0-s11}, {a0-a3}, -160
    53e0:       892e                    mv      s2,a1 <- not covered by PUSH
```

then is it feasible to reassign s2->s1 later? as a peephole optimisation? I'm not a compiler engineer so I don't know the consequences on the compiler, but reallocating that one register would improve the code size.

tariqkurd-repo, 2021-10-06:
Hi @kito-cheng 

This is the PR for the PUSH/POP implementation.

https://github.com/plctlab/llvm-project/commit/fa131e98fb08ca957110fc9008e95a0d086e96cf 

It has `areg_list` excluded as it was not clear how to implement it.

Your input and also from @topperc says to me that it *is* possible to implement it, and the complexity will come from trying to optimise the register usage, so that more of the `mv sX, aX` moves can be included in the `push` instruction.

In the future it may be possible to find a way of adjusting the register allocation to make the moves fit better.

Long term code-size improvement is the goal, and I don't want to limit the spec because of current compiler limitations. I think the conclusion is that `areg_list` should be included,  in the `push` spec, and so that in the future compiler experts can improve the implementation of LLVM. Do you agree?

@jim-wilson what's your opinion on the GCC implementation of `areg_list`?

topperc, 2021-10-06:
Maybe we could visit every instruction in the function and record all defs and uses of two s registers then go back through the function and swap them? Though that's probably not always safe. One thing that would make it not safe is https://gcc.gnu.org/onlinedocs/gcc/Local-Register-Variables.html#Local-Reg-Vars. If a variable is forced to a specific register we can't change it since it might be used by inline assembly.

kito-cheng, 2021-10-07:
> If a variable is forced to a specific register we can't change it since it might be used by inline assembly.

GCC's regrename has handle that case[1], no rename for hard register used in inline asm, I guess we could did similar check in LLVM? that's definitely an issue we should handle very carefully when implement that. 

[1] https://github.com/gcc-mirror/gcc/blob/master/gcc/regrename.c#L1785

jnk0le, 2021-10-07:
s0 and s1 reside in compressed range, so it might turn out that moving there different arguments will provide better compression.

tariqkurd-repo, 2021-10-08:
>s0 and s1 reside in compressed range, so it might turn out that moving there different arguments than a0/a1 will provide better 
>compression.

c.mv supports moving all registers, so the 8-register set is not relevant to this (if I understand your comment correctly)

tariqkurd-repo, 2021-10-14:
Thinking about this further, if the compiler doesn't support _areg_list_ then code-size will suffer but it doesn't affect the specification of the instruction, because the _areg_list_ moves simply become redundant. All s registers are saved before they are overwritten, therefore if the compiler wanted the value in the s register (overwritten by the move) it would have to load it from memory again anyway (otherwise it wouldn't have bothered to spill it in the first place).

The intention is to keep _areg_list_ in the spec as for some codes it can save > 1% of code size, and the compilers should have at least the simple implementation of deleting any moves which are redundant because they were included in the _mv sX, aX_ moves. The more advanced implementation of reallocating s registers so that more moves can be deleted should also be completed, but it more complex and will take longer. This is one area where compilers can differentiate themselves as the potential gain of an optimal solution to this problem is large.

closed 2021-10-14 by tariqkurd-repo