riscvarchive/riscv-code-size-reduction #122: Interrupt safe sequences in CES are not interrupt safe.

RogierBrussee, 2021-11-29:
The CES spec draft is using an "equivalent interrupt safe asm sequence" to explain the semantics of push. E.g.

c.push {ra, s0-s1}  -32

in RV32I should correspond (which I understand means "can be implemented as expanding to the following sequence of 32 bit instructions and should perhaps be formulated as such)

addi sp, sp, -32
sw  ra, 28(sp) 
sw  s0, 24(sp)
sw  s1, 20(sp)

However this sequence is _not_ interrupt safe. If the sequence is interrupted, say at the sw s0, -24(sp) instruction, when the the sequence is restarted from the beginning, the sp register will be decremented twice. I think the interrupt safe sequence should be

sw  ra, -4(sp) 
sw  s0, -8(sp)
sw  s1, -12(sp)
addi sp, sp, -32

with similar adjustments made to the other examples. 

I don't see c.pop in the pdf, but the the sp adjustment must also come last for pop just like for "usual" code generation.

I don't see a c.popret description in the pdf either but for c.popret there seems to be a problem:

ADDI sp, sp, imm
# If an interrupt occurs here and we restart the sequence, the sp gets incremented twice
JR ra

I don't see  how there can be an equivalent restartable sequence unless there is only one non idempotent instruction in the sequence which is the last one.  To make a c.popret instruction have an equivalent restartable sequence something like an indivisible "Jump Register And Add Immediate" instruction i.e. an hypothetical primitive instruction

JRAADDI rd rs1 rs2 imm     # indivisible  (pc, rd)  = (rs2, rs1 + imm) 

somewhat similar to 

JALR rd rs1 imm  # indivisible (pc, rd) = (rs1 + imm, pc)

seems to be needed . The problematic sequence can be replaced with an indivisible

JRAADDI sp, sp, ra, imm

Of course only special cases of the hypothetical JRAADDI instruction are used. Put another way, for an equivalent sequence for c.popret that one can expand to, I think one only needs a _guaranteed_ fuse of 
 
C.ADDISP  sp, imm
C.JR ra

That one can treat as a primitive indivisible instruction.

tariqkurd-repo, 2021-11-29:
HI @RogierBrussee ,

The interrupt safe equivalent sequences are what the compiler generates now, which is what we are replacing with PUSH/POP.
Therefore the restart point is specific to the instruciton which is interrupted, it doesn't restart from the beginning. The point is to show what compiler genrated output is being replaced.

closed 2021-11-29 by tariqkurd-repo

tariqkurd-repo, 2021-11-29:
@RogierBrussee are you a RISC-V member? We had a discussion about how to represent this in the spec on the mailing list quite recently, so you can see the details there.