riscvarchive/riscv-code-size-reduction #123: Change JVT to Supervisor CSR?

David-Horner, 2021-11-30:
The intended use of the jump table is to provide an efficient alternate reduction for R_RISCV_CALL and long jump for high use calls/jumps. It provides an optimized PLT.

A M/S/U system has the virtual memory support to make the table read-only. 
Further a M/U embedded system can make the table read-only [or read execute] with MPM entries. 

However, if JVT is writeable by user-mode process, it can defeat these measures and hi-jack the mapping tables by replacing with its own.

There are of course many methods to limit the impact, 
e.g.1)  requiring the table to be accessed from a currently read-only memory area
    .2) provide a lock/unlock facility for S mode.

However, I believe the simplest is to make JVT a supervisor table.
The linker/loader will request a table from the O/S just as it requests storage allocation.
If the O/S refuses then there is no table for the linker/loader to use.

tariqkurd-repo, 2021-11-30:
@aswaterman and co. are currently revewing the spec. Tagging him so he knows about your input @David-Horner

aswaterman, 2021-11-30:
I think this attack is only possible if the attacker already has found a gadget to jump to an arbitrary address, as the attacker would need to force a jump to an instruction that loads JVT.  So would this really improve security in practice?

Also, note that the same argument (and counter-argument) also apply to gp, since function pointers in the data section can be accessed with gp-relative addressing.  And of course gp is always writable in userspace.

aswaterman, 2021-11-30:
I'll also cc @jhauser-us since he's been liaising between AR and the code-size group more than me.

jhauser-us, 2021-12-01:
The ability for user code to change `jvt` is qualitatively the same risk as letting user code modify x1 (ra), or any other x register.  Quantitatively, the risk with `jvt` is actually much lower than those other registers, if we can assume there are only a few instructions in the entire program that set `jvt`.  I would expect often there would be only one such instruction, during program start-up, making `jvt` a much, much smaller target for finding exploitable "gadgets" to set it than for all the regular x registers.

tariqkurd-repo, 2021-12-02:
Yes I agree with both @aswaterman and @jhauser-us . There will be so few instructions which program JVT that a gadget is very unlikely. it was a good point, and well worth raising.

closed 2021-12-02 by tariqkurd-repo

dansmathers, 2022-09-05:
Would it make sense to use the mseccfg register like is used for permissions to use the user-level crypto seed csr?  It gives security the option to lock things down.  seems analogous.   It feels weird that the vector table can be execute only (no read/write permission) but the jvt CSR is still read/write.