riscvarchive/riscv-code-size-reduction #124: Make zero config in JVT disabled and 000001 be vectored.

David-Horner, 2021-11-30:
Using 0 in the CONFIG field in sjvt allows simple management in S mode.
This follows the convention for satp.MODE.
Similarly, mtvec.MODE = 1 is a "vectored mode" whereas 0 is not [it provides a single handler entry]

CLIC is developing a vectored mode for interrupts that is effectively the same as the base mode for JVT, that is a table of virtual entry point addresses. In that case they currently use mtvec.MODE = 3 to enable the new  **x**tvt CSRs. CLIC can justify a new set of CSR and use an existing M-mode CSR to flag the mode. 
Do we have a similar luxury? Should there be 3 JVT registers? One each for control in M, S and U? 
I think the base mode having only one in S supporting M,S and U is sufficient.
This is a deviation from earlier CSR allocation philosophy, but we are now seeing many of those implicit conventions being abandoned.

tariqkurd-repo, 2021-11-30:
@aswaterman and co. are currently revewing the spec. Tagging him so he knows about your input @David-Horner

tariqkurd-repo, 2022-02-08:
This has been reviewed by @aswaterman and @jhauser-us and they are happy with the tagged spec release v0.70.1 where jvt.mode=0 is table jump mode. I believe that consistency with the CLIC has been considered.

closed 2022-02-08 by tariqkurd-repo