riscvarchive/riscv-code-size-reduction #125: How explicit is the actual store and load order in push and pop?

David-Horner, 2021-11-30:
LLVM recently revised its code to ensure ra would be loaded as far in advance of its use in a ret as possible.
This is because certain micro architectures would not read ra for a ret until it was written to the register potentially a long way down the pipeline.

We can follow suit in the popret instructions , explicitly loading ra as the first register.

However, alternatively we can allow an implementation to effect the same result allowing for memory access reordering.

I suggest we do the latter. As long as the net result is the same the order of the writes and reads, themselves, can change.

This is consistent with how many microarchitectures optimize their memory accesses. As long as they are consistent with RVWMO the actual issue order is mutable.

tariqkurd-repo, 2021-11-30:
I agree and this is already in the spec

_Because the memory is idempotent and the stores are non-overlapping, they may be reordered, grouped into larger accesses, split into smaller access or any combination of these._

The actual hardware implementation is up to the implementation.  Any stores can happen multiple times before completion but the stack pointer update and register moves (for c.pusha) can only happen once.

closed 2021-11-30 by tariqkurd-repo

David-Horner, 2021-12-01:
Even so, it would be valuable to include the suggestion that for popret the ra load occur early in the sequence.