riscvarchive/riscv-code-size-reduction #126: Consider reserving space for a larger table jump table in the encoding

tariqkurd-repo, 2021-12-07:
Table jump (c.jt/c.jalt) have an 8-bit index into the jump table of function addresses.
It's possible that the table should be extended for use by an outliner, and the spec of a possible instruction to call outlined routines has not yet been defined.

Before we fix the encoding we should consider reserving an extra bit of the table index, to allow the table to be extended into a new opcode in the future. This could be for outlined routines, or for anything else.

tariqkurd-repo, 2021-12-07:
@aswaterman @jhauser-us any comments on this one?

David-Horner, 2021-12-07:
CSR xjvt can control the tables future features.

Just as mtvec controls the table for interrupts and has been expanded to allow vectored interrupts.
And with CLIC in riscv-fast-interrupt TG the mtvec is to be expanded further.

The low 6 bits of xjvt CR can be used to reconfigure the split.
It can be reconfigured to even provided tie in to satp ASID or call originating domains.
The options are plentiful with the 6 bits, and those bits can be expanded by using an escape sequence to involve the low 10 bits for a 1K aligned table [or 12bits for a page aligned table].

Given the inherent flexibility of xjvt I don't see value reserving any bit within c.jt/c.jalt, but rather the initial allocation of those bits could be more flexible as I proposed on the mailing list.

David-Horner, 2021-12-16:
Further.
1) the vectoring address bits can be used for per target configuration bits.
     The lowest bit is already "reserved" as instructions have to be 2 byte aligned.
     Given that load module placement and thus entry point addresses is determined by the loader, the number of lower bits can be increased.
    Likewise, feature expansion can use the high bits for embedded systems that know ranges of addresses that are 

2) If the original post is suggestion reserving a bit within the C.jalt/C.jt instructions, I see no value in diminishing the range for the  initial release of these instructions. If an encoding bit is available latter, and the expansion capabilities in the csr and vector table address bits are insufficient, then new variants of C.jalt/C.jt can compete with other candidates.