riscvarchive/riscv-code-size-reduction #129: Do we need auto-compressing of instructions like MUL in MC layer?

luke957, 2021-12-09:
For now, instructions such as NOT, NEG can be auto compressed in MC layer. Do we need to auto compress instructions like MUL?

abukharmeh, 2021-12-09:
IMHO, all instructions that has direct equivalent should be mapped in MC layer !  
I dont see any advantage of mapping them in IR, unless IR has some kind of cost model, that would make it more likely to emit them, @tariqkurd  what do we think ?

abukharmeh, 2021-12-09:
@jeremybennett  Hi Jeremy, any comments on this one.
PS: this is related to https://github.com/plctlab/llvm-project/issues/30

luke957, 2021-12-12:
> IMHO, all instructions that has direct equivalent should be mapped in MC layer ! I dont see any advantage of mapping them in IR, unless IR has some kind of cost model, that would make it more likely to emit them, @tariqkurd what do we think ?

Yes, I think so. For MUL instruction, I understand it's a two-instructions-into-one transformation. So it seems no direct equivalent exists. Should this kind of instruction be considered to be auto-compressed in MC layer?

abukharmeh, 2021-12-12:
Do you mean muli (li + mul), Yes that's 2 instructions into one? or do you mean c.mul which would map into a normal mul ?  

Perhaps we can create a list of all instructions and see what could be done where, and then we can ask for input on the mailing list if anybody has any suggestions or objections? (I think there are a couple of compiler experts on the list).

I am not completely sure, but I would imagine mapping new instructions into already existing instructions (e.g in MC) would likely be the path of least resistance when trying to push code into the mainstream? and would offer the same code savings, unless there is deliberate targetting for compressed instructions in IR.

abukharmeh, 2021-12-12:
I just had a look into the new commits, and its appears that everything that can be done this way is already done I think

zext, sext and mul: https://github.com/plctlab/llvm-project/commit/d47f6114645a2e73687ad58343709deaff991246
not and neg: https://github.com/plctlab/llvm-project/commit/ce42f8fe042052a02f48f40f8867c91ea2123746
lbu, lhu, lb, lh, sb, sh: https://github.com/plctlab/llvm-project/commit/3ce48744146620d8f951ea9814acf4081534e251

So likely you now asking if we should auto fuse instructions at assembly level (e.g MULI) ?

luke957, 2021-12-13:
Yes, I mean muli (li + mul). Should we auto fuse instructions at assembly level? That might be a little bit complicated.

closed 2022-06-28 by abukharmeh