riscvarchive/riscv-code-size-reduction #130: incrementing vs. decrementing address order in PUSH/POP instruction sequences

jeras, 2021-12-20:
I already reported a similar problem with the GCC compiler a few years ago.
https://github.com/riscv-collab/riscv-gcc/issues/132

An incrementing address sequence could have better performance than a decrementing sequence.

## Data cache and burst read example

As an example I would like to use a data cache with a few specific features, this might not apply to all cache implementations.
1. On a load cache miss, this cache executes a **wrapped burst access starting with the requested address**. The burst goes toward the main memory or next level cache.
2. Here I will assume the burst and cache line sizes are 64B and the upstream data bus width is 32-bit or 4B (32-bit RISC-V CPU).
3. The cache can return the requested data to the core as soon as the first word in the burst is returned. The next word in the cache line is available to the CPU in the next clock period without any extra delay.

Now for a case where an incrementing POP would have a much better performance than a decrementing POP.
A. For an **incrementing** POP sequence, there would be an initial latency before the first load is completed (waiting for the first word in the burst is returned). There would be no latency for any of the following loads, as long as the loads address the same cache line.
B. For a **decrementing** POP sequence, the latency for the first load would be the same as in the incrementing case. The second load would have to wait for the burst to complete before the cache could return the data to the core. All other loads in the sequence would complete without additional delay.
The **latency** for the entire POP sequence is 64B/4B-1=**15 clock periods longer** (the length of the burst in clock periods) in the decrementing case compared to the incrementing case.

I did not look into specific RISC-V implementations of data cache controllers and burst access to main memory, so somebody with experience there might comment. The last time I looked at caches was fixing a somehow related performance issue in an OpenRISC based SoC with a single level of caches and a SDRAM controller.

## Tightly coupled memory (TCM) with a 2 clock period latency for read access example

Another example is where the TCM has a 2 clock period latency due to a register cutting a long timing path either on the address or the read data side. The throughput of such a TCM can be increased with a simple scheme, by prefetching address+1 on each load. The first random access would take 2 clock cycles, following accesses with an incrementing address would take 1 clock cycle.

An **incrementing** POP would incur an 1 cycle delay on the first load in the sequence, the following loads would be executed in a single clock period. A **decrementing** POP would incur a 1 cycle delay on each load in the sequence, thus almost **halving the throughput**.

## Summary

The RISC-V instruction set does not have a requirement for individual stack accesses loads to be done in decrementing order, since the stack pointer is not modified by the same instruction as the loads. The downward growing direction of the stack is entirely defined by the instruction modifying the stack pointer.

I did not research if there are any serious backward compatibility issues to address.

tariqkurd-repo, 2022-01-17:
Hi @jeras , sorry I only just spotted this issue.
The implementation is free to map PUSH/POP to incrementing bursts. The decrementing order in the spec is only an example of an implementation - the load/stores can be reordered so that incrementing bursts can be used. The spec does say that the load/stores can be issued in any order and merged into larger accesses or split into smaller accesses.

I hope that clears up the problem.

Tariq

closed 2022-01-17 by tariqkurd-repo

jeras, 2022-01-17:
Do you se any need for compatibility between PUSH/POP and sequences generated by the compiler?
For example linking together code compiled with PUSH/POP and code compiled without PUSH/POP.