riscvarchive/riscv-code-size-reduction #133: Debug trigger on target_address

Silabs-ArjanB, 2022-01-17:
In the Zc* v0.61.0 specification the descriptions of both c.jt and c.jalt each contain two if statements based on debug_trigger(target_address). This to me does not seem consistent with the handling of RV32I jumps or branches in which the 'target_address' never ends up in DPC or MEPC.

The checking of the final 'target address' is part of the responsibility of the next instruction (which might experience an instruction access fault, debug_trigger with timing=0, etc.). This issue is similar to the one raised earlier in https://github.com/riscv/riscv-code-size-reduction/issues/76.

tariqkurd-repo, 2022-01-17:
Hi @Silabs-ArjanB - what I was trying to do here was to separate triggers from the table_address and target_address so they can be handled correctly.

Here's the code with a number after each if statement:

```
  if (debug_trigger(table_address) && MCONTROL.timing==1 && MCONTROL.action==1) { // 1
    DPC        = current_PC;
    DCSR.cause = 2;
    enter_debug_mode();
  } else if (debug_trigger(table_address) && MCONTROL.timing==1) { // 2
    MEPC       = current_PC;
    MTVAL      = table_address;
    MCAUSE     = BREAKPOINT;
    take_exception();
  } else if ((debug_trigger(target_address) && MCONTROL.timing==0 && MCONTROL.action==1) { // 3
    DPC        = target_address;
    DCSR.cause = 2;
    enter_debug_mode();
  } else if (((debug_trigger(target_address) && MCONTROL.timing==0) || // 4
               !can_access_instruction_memory(target_address)) {
    MEPC       = target_address;
    MTVAL      = target_address;
    MCAUSE     = debug_trigger(target_address) ? BREAKPOINT : INSTRUCTION_ACCESS_FAULT;
    take_exception();
  } else {
```
1: debug mode entry on table_address
2: exception handler entry on table_address
3. debug mode entry on target_address
4. exception handler entry on target_address

are you suggesting that I remove 3 and 4, as they are covered by the execution of the target instruction>?

Tariq

Silabs-ArjanB, 2022-01-17:
Hi Tariq,

Yes, that is exactly what I am suggesting.

Best regards,
Arjan

tariqkurd-repo, 2022-01-17:
OK, agreed. I've updated the spec and moved the version to 0.61.1.

closed 2022-01-17 by tariqkurd-repo