riscvarchive/riscv-code-size-reduction #134: Handling of dereferencing inconsistent with CLIC spec

Silabs-ArjanB, 2022-01-17:
The table jump instructions (c.jalt and c.jt) and the CLIC's hardware vectoring mechanism (see https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc#new-xtvec-csr-mode-for-clic) are very similar (they both jump to a pointer obtained from a table).

The retrieval of the pointer from the table however seems to be treated differently in both specifications and it would be nice if this can get aligned.

The Zc* specification states:

> The memory pointed to by JVT.base is treated as instruction memory for the purpose of executing table jump instructions.

CLIC's hardware vectoring mechanism specifies the following (see above link):

> For permissions-checking purposes, the memory access to retrieve the function pointer for vectoring is treated as a load with the privilege mode (also obeying MPRV and SUM bits) and interrupt level of the interrupt handler. 

To me (please correct me if I am wrong) the Zc* specification therefore implies that 'execute permission' is needed to read from anywhere in the jump table, whereas the CLIC specification implies a check with R/W bits (if you have a PMP).

I would greatly prefer if both specifications can get aligned on this (or to hear an explanation why both specifications should differ here). 

I prefer the Zc* specification with respect to this topic and I will therefore post the same question as an issue in the Fast Interrupts (CLIC) TG.

Silabs-ArjanB, 2022-01-17:
FYI: I put the matching issue for CLIC here: https://github.com/riscv/riscv-fast-interrupt/issues/191

kasanovic, 2022-01-18:
The fast interrupt vector table treats table lookup as a load so that MPRV mechanism can be used for emulation of table load if needed.  Also, the table is usually written/read with load/store instructions.

tariqkurd-repo, 2022-01-19:
OK, I get it, so MPRV can be used to emulate code running at a lower privilege level.

For table jump I wanted to express it so that it can all be executed in the instruction fetch, it doesn't need to use a load for jump table access, so that table jumps can be executed early in the pipe. 

How does this affect the specification? After all the instruction fetch can make use of MPRV for the jump table access if necessary, although this would seem irregular, but then it's much better if the two specifications match. In the CLIC you have:

`pc := M[TBASE + XLEN/8 * exccode] & ~1`

this needs read access, not execute, as it is a load, and so it will be loaded into the D$. A core implementing S-mode and table jump would need to cache the TLB read permission bit in the instruction fetch, and only check it for this one purpose, but will also load the table into the "wrong" cache which may cause issues.

Allowing table jumps to be executed in the instruction fetch alone is performance critical, because they will execute incredibly frequently (i.e. almost every function call is a table jump), so we can't afford to wait for load access each time. Interrupts clearly occur much less frequently.

Given that the main customer for table jump is small embedded cores which typically only implement M/U mode then the difference is probably minimal, but the spec needs to be correct allow S-mode to be implemented, and usable.

Any advice about what to do? What about emulation support for table jump? It's better if it is supported.

tariqkurd-repo, 2022-01-19:
Hi @kasanovic - yes Sbare might be supported, I should have said if VM or PMP is implemented, as PMP has all the same concerns.

Looking at the jump table permissions:

1. If the jump table only has execute permissions, then it can't be scanned for gadgets. 
2. If the jump table has read-and-execute permission then it can be scanned for gadgets, and executed from. 
3. If the jump-table has read and not execute permission then we can't I-fetch from it, but we can "special case" this, as in the CLIC.

If the jump-table requires read access, then it may not be in the same VM/PMP region as the rest of the code, assuming that some security is in place, which would be execute only.

I think it's reasonable to have the jump table in a data section, as that will allow software to have the option of updating it.

_>You can execute table jump in the instruction fetch pipeline without allocating the table entries in the data cache, and while checking for correct permissions for a load._

This is ok for me and will give consistency with the CLIC.

If this all sounds ok I'll update the spec to say that load permissions are checked, not isntruction fetch permissions.

Silabs-ArjanB, 2022-01-19:
Hi @tariqkurd-repo ,

> If this all sounds ok I'll update the spec to say that load permissions are checked, not instruction fetch permissions.

With 'instruction fetch permission' do you mean to say 'execute permission'? 

Apart from the permissions the other thing that differs between CLIC and Zc is the usage of MPRV for the access into the table itself. Are you now planning to add that as well?

tariqkurd-repo, 2022-01-19:
_>Apart from the permissions the other thing that differs between CLIC and Zc is the usage of MPRV for the access into the >table itself. Are you now planning to add that as well?_

If we change it from "instruction fetch permission" to "load permission" then it will have to behave as if it is a data load, following whatever rules are implements for loads.

So really we need to treat it as one type of memory or the other and choose whatever makes most sense.

kasanovic, 2022-01-19:
>>>>> On Wed, 19 Jan 2022 07:53:00 -0800, Tariq Kurd ***@***.***> said:

| Hi @kasanovic - yes Sbare might be supported, I should have said if VM or PMP
| is implemented, as PMP has all the same concerns.

| Looking at the jump table permissions:

|  1. If the jump table only has execute permissions, then it can't be scanned
|     for gadgets.

Scanning at run time isn't the only way to find gadgets.
Memory image can be known ahead of time by attacker.

|  2. If the jump table has read-and-execute permission then it can be scanned
|     for gadgets, and executed from.
|  3. If the jump-table has read and not execute permission then we can't
|     I-fetch from it, but we can "special case" this, as in the CLIC.

| If the jump-table requires read access, then it may not be in the same VM/PMP
| region as the rest of the code, assuming that some security is in place, which
| would be execute only.

| I think it's reasonable to have the jump table in a data section, as that will
| allow software to have the option of updating it.

Right - at least for the CLIC, there are certainly use cases where the
vector table entries are updated at runtime, so would need data write
permissions also, which would also not allow to be in same permissions
region as code.

Krste

jhauser-us, 2022-01-20:
The only argument I've found very convincing for preferring read permission instead of execute permission is this one from Krste:
> Another minor concern with giving table entries execute permissions is that they are arbitrary function pointers, which can now be jumped into and executed as code, possibly allowing some more gadget construction.

But then it occurred to me, it's already possible to jump into the middle of an instruction, at the second halfword of a 4-byte instruction, which I think provides a comparable ability to execute unintended bytes as instructions.  Depending on how lucky/unlucky you are, who knows how many 4-byte bogus instructions could be executed in sequence before instruction fetch manages to lock back to the intended instruction boundaries.

Furthermore, it's not like jumping arbitrarily into the middle of compiled code is somehow preferable, even if you do it at a proper instruction boundary.  Why should I believe that's any better than jumping into the middle of data (in this case, an array of code addresses) and executing whatever that decodes into?

One advantage I see to requiring execute permission for the table lookups of JT and JALT instructions is that it's more consistent for code that is execute-only, not readable.  The jump table is part of that code, so ideally it should be equally hidden, not readable.

I'm afraid I can't see why I should expect machine-level code to need to emulate table-jump instructions.  Krste wrote:
> The fast interrupt vector table treats table lookup as a load so that MPRV mechanism can be used for emulation of table load if needed.

Even if we accept that it's necessary to emulate a read of the jump table, a problem occurs only if the table is execute-only, not readable.  But in that case, it's very likely the rest of the code is also execute-only, and machine level won't be able to read the actual JT or JALT instruction either, to begin the process of emulating it, in any scenario where the instruction isn't conveniently provided in `mtval`.  Right?

As Tariq has noted, execute permission favors (weakly?) doing the table lookup in the front end of a processor's pipeline, perhaps as a special form of instruction fetch.  I'm pretty sure this will usually be the simpler technique for smaller cores, though it may be less favorable for high-performance cores.  But given that we expect small cores to be the primary implementors of these instructions, not high-performance cores, I think it's right to give preference to their biases.

All in all, so far, I see the arguments favoring execute permission, not read permission.

tariqkurd-repo, 2022-01-20:
>Right - at least for the CLIC, there are certainly use cases where the vector table entries are updated at runtime, so would need data write permissions also, which would also not allow to be in same permissions region as code.

We don't have this for table jump - different tables are possible for different contexts, but these are managed by switching the JVT CSR to the new location. I don;t expect the table to actually require any management (and if it does then of course the OS can set W permission to the page).

> One advantage I see to requiring execute permission for the table lookups of JT and JALT instructions is that it's more consistent for code that is execute-only, not readable. The jump table is part of that code, so ideally it should be equally hidden, not readable.

Very true - it's a common mode problem.

> All in all, so far, I see the arguments favoring execute permission, not read permission.

I think this is the simplest approach, certainly for embedded cores as you say John.

If there are no compelling arguments to change, then I'll leave the spec as it is.

tariqkurd-repo, 2022-01-21:
closing as I think the current spec is ok - I'll re-open if necessary

closed 2022-01-21 by tariqkurd-repo