riscvarchive/riscv-code-size-reduction #139: Irregularity for register s10/s11 in push and pop

RogierBrussee, 2022-01-26:
The handling of s10 and s11 in push and pop is irregular in that s10 and s11 come as a package deal. This permeates irregularity in things like computing the stack_adj_base which has to be done by hardware and is now a big switch in the spec rather than stack_adj_base = top(#registers in the registerlist * XLEN/(16*8))  *16  ==  rlist[2:3] *16 for RV32; (rlist[1:3] -1)*16 for RV64 .  However register s11 seems little used statically (see figure 5.1 in Waterman's thesis https://escholarship.org/content/qt7zj0b3m7/qt7zj0b3m7.pdf).  

What is the cost of being only able to push/pop {ra, s0-s10} with CM.PUSH/POP instructions (i.e. up to 12 registers) and requiring an additional C.SWSP/C.LWSP s11  in the rare case that s11 needs to saved/restored?  

P.S. One of the reasons s11 is saved/restored is that _all_ registers are saved/restored in an interrupt handler. Maybe a pushall/popall instruction is worth an irregularity for rlist==15. But then, if pushall/popall instructions have any use, maybe a separate opcode (requiring no registers and no immediate) is simpler, more useful and cheap if you can already do push and pop. 

P.P.S. Since there is no RV64E there does not seem to be a problem to have rlist run from 2  upto and including 14 on RV64  which would allow push/pop of s11 and make stack_adj_base = rlist[1:3]*16.

RogierBrussee, 2022-01-26:
Hi @tariqkurd-repo,

thanks for your answer.  
>yes, storing {ra,s0-s11} is unusual but is more frequent than {ra,s0-s10} in the benchmark suite.

I can believe that: and it seems to point to the main usage of push/pop ing s11 being push/pop ing all registers. 
I had also already understood that from the spec.

>There was previously discussion about pushall/popall functionality, but this full register list is the closest we come to it.

You probably understood this already, but just to be clear:  I am not suggesting an instruction that has an rlist with between 1 and 31 of registers, but one with only an opcode, (say like EBREAK) with no rlist and no immediate, that pushes all 31 registers except x0 on  the stack and adjusts the sp .by 8*16 (RV32) or 16*16 (RV64). But you probably already discussed this.  

But if the main reason for push/pop  {ra, s0-s11} is to get almost halfway to save/restore all registers,  then it should be accompanied by another 31 - 13 = 18 LWSP, 18 SWSP + 2 ADDISP,  Changing it to 19 LWSP, 19 SWSP + 2 ADDISP does not seem to make the big difference. 

>We have 12 options so we can reserve four for the .E version, so we can have simple decoding of the two opcodes. I don't >know how many .E encodings we actually need yet as the EABI isn't tied down.

OK. We don't have to do all the options in one compressed instruction though. 

>What are you concerned about? Complexity in the hardware implementation?

There is no question it is doable. My concern is that the added complexity in the decoding phase does not seem to bring real benefits  The irregularity of ra and s0-s1, vs s2-s11 is clearly justified. However, push/pop {ra, s0-s11} should be pretty uncommon to begin with, and if it is used, it should be either in an interrupt handler and you must save and restore an additional 19 instead of 18 registers, or you should have a long complex function with (probably) 7+ arguments and push registers went up from 16 to 2*16 bit. Down from 14*16.
Introducing irregularity for that seems to be the wrong tradeoff with KISS, and to be honest, it just looks out of style between the other RV instructions.  

>It seems that setting #registers_in_the_register_list to 13 in this case seems pretty minor compared to everything else >needed to implement push/pop.

It is the difference between 

stack_base_adjust = rlist[2:3]<<4 (RV32)
and
stack_base_adjust = (rlist =! 0b1111)?(rlist[2:3]<<4) : 13

sure can be done, but it does not feel "elegant" (unlike the first version where when I wrote it I thought: that is a neat! The spec reserves code for the E-extension _and in the process_ simplifies the decoding of stack_base_adjust ... except for that irregularity) . But sure, can be done. 

>If we exclude {ra,s0-s11} then we need to double the code-size in the case where it's needed - c.swsp is as big as cm.push.

You go from one to two 16 bit instructions in the rare case that you push 13 registers. I doubt it would even register in a benchmark for the reason I explained with my concern. But I don't have the benchmarks to prove it 

>Overall we've gone for best code-size, as that's our main objective. 

Ok.  

>Performance loss / stack memory increase from >including s11 in the list is taken as less important.

The performance loss is an additional concern but I can see the trade off there. 

Best,

Rogier

closed 2022-01-26 by RogierBrussee

tariqkurd-repo, 2022-01-27:
The pushall/popall was discussed by the interrupt controller (CLIC?) group sometime ago and the conclusion was that the benefit wasn't high enough, I had proposed something like that in an early version of this spec. 

The register list {ra,s0-s11} does happen more often than you might think, which surprised me. Having the whole thing in one register list is more efficient for the implementation, not least because you know that the whole memory block will be accessed so you can optimise the cache/bus access.

But yes - it does cause that one irregularity - hopefully the overall benefit will help us in the end.

RogierBrussee, 2022-01-28:
Hello Tariq,

Thanks for your explanation!

Rogier