riscvarchive/riscv-code-size-reduction #141: non destructive compressed shifts?

jnk0le, 2022-02-04:
single direction on RV32 needs the entire 11 bits of opcode so it's a heavy encoding, but maybe worth to consider wrt Zba and Zbb that diminished some of use cases for shifts (especially destructive ones).

tariqkurd-repo, 2022-02-04:
sorry - what are you suggesting? Can you give me a clear example?

jnk0le, 2022-02-04:
RV32 only left or right (or both) shift as `cm.*` instruction with

`rd` in [4:2]
`rs1` in [9:7]
`uimm5` in [12:10] and [6:5]

Currently available shifts are destructive only, overlap with them handled similarly to `c.lbu` and `cm.lbu`.

tariqkurd-repo, 2022-02-07:
it's not clear this will help reduce the code-size enough, I'll reopen it if there's evidence that it will make a big difference