riscvarchive/riscv-code-size-reduction #144: Zc*, misa.C and ISA strings

Silabs-ArjanB, 2022-02-09:
Hello, 

What is the relationship between the Zc* extensions and the misa.C bit?

The thing that makes it not obvious is that the C extension is stated to be Zca + Zcf. (Also the fact that C is specified in the RISC-V unprivileged spec to depend on F and/or D make the rules and naming non-obvious)

So for example if I have a CPU supporting amongst others Zca + Zcf does that imply misa.C = 1? (I would guess not as I would think that only if C is part of of the ISA string that misa.C = 1).

This leads to another question: Should or can 'C' be part of the ISA string if any of the Zc* extensions are also part of the ISA string? For example, is RV32IMACZcmp legal (I would think so), is RV32IMACZca_Zcmp legal (I guess not), is RV32IMAC equal to RV32IMAZca_Zcf (implying also equal misa.C bits), is RV32IMAC_Zcmp equal to RV32IMAZca_Zcmp, etc.?

Best regards,
Arjan

tariqkurd-repo, 2022-02-09:
good question. @aswaterman @jhauser-us - any comments on this?

aswaterman, 2022-02-10:
The misa.C bit is defined in terms of the already-ratified C extension.  Nothing the code-size reduction TG produces will have any impact on the definition of that bit.

Zca/Zcf aren't ratified extensions, so I can't provide a definitive answer to that question.  What I can say is that cores that implement all of the instructions in the C extension should set the misa.C bit.  If corresponding F/D extensions are omitted, cores should still set the misa.C bit even though the corresponding FP loads/stores are omitted.

tariqkurd-repo, 2022-02-10:
I think I'd expect misa.C to mean that the core supports 16-bit encodings, and the minimal set is Zca.
@aswaterman shall I put something in the spec about this?

aswaterman, 2022-02-10:
@tariqkurd-repo IMO, adding a statement to the spec will confuse matters more than anything. The spec’s definition of misa.C is already unambiguous, and it’s hard to see how a redundant restatement of it will improve the situation. As a general sentiment, we should factor out common architectural principles and avoid repeating them.

linsinan1995, 2022-03-03:
Hi @tariqkurd-repo , I have a similar question on the ISA string part. E.g. if the ISA is rv32gc_zca, it is an illegal combination or we disable FP instruction in c-ext? Could we include this part into the spec?

tariqkurd-repo, 2022-03-07:
Hi @linsinan1995 I think that's an illegal combination - it should be rv32g_zca, as zca is a subset of C you can't have both.
@aswaterman is that what you expect?

aswaterman, 2022-03-08:
It's the other way around - it's OK to specify overlapping extensions in the same ISA string.  rv32gc_zca is legal and means the same thing as rv32gc.

Re: @linsinan1995's question: extensions are never subtractive.  Appending something to an ISA string never deletes instructions.

tariqkurd-repo, 2022-03-08:
ok, thanks @aswaterman the clears it up perfectly.

liweiwei90, 2022-03-31:
Are rv32imac_zca_zcmp and rv32imafd_zca_zcmp legal?

tariqkurd-repo, 2022-04-01:
Hi @liweiwei90 , yes they're legal. You can't have C+D and Zcm* as Zcm* reuse the C.FSD*, C.FLD* encodings

So if a configuration string has more than one mapping for a single encoding then it's illegal

Tariq

liweiwei90, 2022-04-01:
OK, Thanks a lot!

> Hi @liweiwei90 , yes they're legal. You can't have C+D and Zcm* as Zcm* reuse the C.FSD*, C.FLD* encodings
> 
> So if a configuration string has more than one mapping for a single encoding then it's illegal
> 
> Tariq