riscvarchive/riscv-code-size-reduction #145: Problem

Xinlong-Wu, 2022-02-11:


abukharmeh, 2022-02-11:
Zc extension is divided into
1. Instructions targeting embedded processors (reuse double floating point compressed instructions):
    a) Double floats are not often used in embedded space. 
    b) Lack of encoding space for 16 bit instructions
2. Instructions targeting both embedded as well as and applications processors (uses unique encoding)

tariqkurd-repo, 2022-02-11:
Hi @Xinlong-Wu what do you think is incompatible?

Xinlong-Wu, 2022-02-12:
> Hi @Xinlong-Wu what do you think is incompatible?

I mean, i have seen sentence like that in many inst set (zcmb, zcmp, zcmpe, zcmt)
[image]

the C ext don't have any subset, that means we need to disabled entire C ext for these inst set.

while, I think it is worth to be discussing, cuz there's an 16 bit instruction called `c.nop` defined in C ext which is designeded for align the 16bit instructions.

from my previous experience, there should be problem when we compile programs with abi ilp32. because it seems need be aligned 16 bit inst by `c.nop`

type by phone, sorry for my typo.

tariqkurd-repo, 2022-02-14:
Hi @Xinlong-Wu - you need to define Zca as the subset of C which excludes floating point loads and stores. Everything is compatible with Zca. Does that solve the problem?

Xinlong-Wu, 2022-02-14:
I will try. but I think it also should be specified in spec (C or Zc or even both).

because it is not just a problem of how to implement. I don't think we have the right to change the existing implementation of C ext without any spec.

we can solve it later. this issue will not block us now. we are updating the code. it's ok we back to this issue when it should be able to compile progrems.

tariqkurd-repo, 2022-02-14:
HI @Xinlong-Wu - this will all be in the architecture manual when Zc* is ratified. It's the first time we have a subset of C defined, so it is a new concept. I'll close this now but please reopen it if you have further problems.

closed 2022-02-14 by tariqkurd-repo