riscvarchive/riscv-code-size-reduction #148: Zcf requires Zca

tariqkurd-repo, 2022-03-03:
Spec v0.70.2 was missing the text that Zcf requires Zca

tariqkurd-repo, 2022-03-03:
Fixed in v0.70.3

closed 2022-03-03 by tariqkurd-repo

Sanjay-A-Menon, 2022-05-26:
Hi,

Won't Zcf extension with C extension and without Zca extension still be a legal implementation?(though it doesn't make sense, because C is a superset for Zcf and Zca). Plus, why is there a dependency for Zcf on Zca?.

linsinan1995, 2022-05-27:
> Won't Zcf extension with C extension and without Zca extension still be a legal implementation?(though it doesn't make sense, because C is a superset for Zcf and Zca). 

rv32gc_zcf is the same as rv32gc (no new instructions introduced in Zcf). There is some discussion about c and zc* in isa string(https://github.com/riscv/riscv-code-size-reduction/issues/144). 

> Plus, why is there a dependency for Zcf on Zca?.
I don't see any real dependency, maybe Zca is designed as a root for Zc extension? @tariqkurd-repo

Sanjay-A-Menon, 2022-05-27:
@tariqkurd-repo and @linsinan1995, Thanks for the reply. That cleared out a lot of my queries.