riscvarchive/riscv-code-size-reduction #149: ZCF does nothing in RV64?

linsinan1995, 2022-03-06:
c.flw, c.flwsp, c.fsw, and c.fswsp are rv32fc only, so there is nothing to change with zcf arch in rv64?

tovine, 2022-03-06:
You are correct. Zcf doesn't really exist or make sense for RV64 (IIRC).
This should probably be mentioned in the spec, at least with a note - so it doesn't cause confusion regarding `C.LD` and the other instructions that replace them in RV64...

tariqkurd-repo, 2022-03-08:
@aswaterman can you confirm whether rv64_zcf should be illegal, or whether zcf is just defined as empty, so rv64_zcf is equivalent to simply rv64. I suspect this is the first time this kind of thing has happened, so it's better to confirm the behaviour.

aswaterman, 2022-03-08:
I'll punt this one to @jhauser-us

tariqkurd-repo, 2022-03-10:
my preference here is to state that Zcf is illegal, confirmation would be good before updating the spec @jhauser-us.

tariqkurd-repo, 2022-03-15:
ok - no response so let's say that Zcf is undefined for RV64, and so will cause an error if it's used.

closed 2022-03-15 by tariqkurd-repo