riscvarchive/riscv-code-size-reduction #157: Special case of push/pop instruction optimization

Xinlong-Wu, 2022-04-21:
In the process of modifying the push/pop implementation, the following prologue was noticed.

```
addi	sp, sp, -64
sw	ra, 60(sp)
sw	t0, 56(sp)
sw	t1, 52(sp)
sw	t2, 48(sp)
sw	a0, 44(sp)
sw	a1, 40(sp)
sw	a2, 36(sp)
sw	a3, 32(sp)
sw	a4, 28(sp)
sw	a5, 24(sp)
sw	a6, 20(sp)
sw	a7, 16(sp)
sw	t3, 12(sp)
sw	t4, 8(sp) 
sw	t5, 4(sp) 
sw	t6, 0(sp) 
```
in the current version, it will be optimised to 

```
sw	t0, 104(sp)
addi	sp, sp, -16
sw	t1, 52(sp) 
sw	t2, 48(sp) 
sw	a0, 44(sp) 
sw	a1, 40(sp) 
sw	a2, 36(sp) 
sw	a3, 32(sp) 
sw	a4, 28(sp) 
sw	a5, 24(sp) 
sw	a6, 20(sp) 
sw	a7, 16(sp) 
sw	t3, 12(sp) 
sw	t4, 8(sp)  
sw	t5, 4(sp)  
sw	t6, 0(sp)  
cm.push	{ra}, -32
```

I know that there is something incorrect about the current implementation, but I don't know what the correct expectations are

now we have problem

1. cm.push will adjust the stack pointer，so i think it should be put to the start of the prologue.
2. The variable corresponding to the ra register is stored in the middle of the stack. while cm.push doesn't appear to write randomly from the middle of the stack (please correct me if I'm wrong)

so i don't know what should i do to optimize it in this case.

tariqkurd-repo, 2022-04-21:
I'm not sure how that was produced by the compiler. I would say that you shouldn't optimise that case because it doesn't have the correct pattern, so it doesn't work. What do you think?

abukharmeh, 2022-04-22:
I agree with Tariq, It feels like this will be calling some function with args and it want to preserve the original args ? Can you also attach the C/C++ source code,  I wonder how common is it for functions to save A registers like this  (I imagine pretty rare) !

Xinlong-Wu, 2022-04-22:
here is the LLVM IR code. 
```
declare i32 @foo(...)
define void @foo_with_irq() nounwind "interrupt"="user" {
  %call = call i32 bitcast (i32 (...)* @foo to i32 ()*)()
  ret void
}
```
The problem is in code `"interrupt"="user"`

this code marks the function as Exception handling functions
it means the function will do "context swap", it save/restore all regs in prologue/epilogue.

so if we want to optimize by push/pop, we need to reorder the stack.
This is what I plan to do.

Or we can choose not to optimize this case, but I don't think that's a good way.

closed 2022-04-22 by Xinlong-Wu