riscvarchive/riscv-code-size-reduction #161: breakpoint allowed to be set on table vector?

dansmathers, 2022-04-28:
This issue was raised on CLIC TG:
https://github.com/riscv/riscv-fast-interrupt/issues/228
Is it allowed to set a breakpoint on the table vector? If so dpc/dret handling should be specified. If not, it should also probably be specified. I would prefer CLIC and Zc to be consistent.

tariqkurd-repo, 2022-05-13:
Hi - sorry for the slow reply.

I'm in favour of only allowing a breakpoint on instructions, and not within the table. This allows the breakpoint to trigger before or after, and not during. I can add this sentence:

`The fetch from the jump table is not considered valid for breakpoints, as it is not fetching an instruction.`

Would that do it?

allenjbaum, 2022-05-13:
Yea, it's an instruction-type fetch, but not the fetch of an instruction. That is sure to trip someone up.

tariqkurd-repo, 2022-05-16:
For a table jump instruction, the table entry that the instruction selects is considered an extension of the instruction itself.
Hence, the execution of a table jump instruction involves two instruction fetches, the first to read the main instruction (_cm.jt_
or _cm.jalt_) and the second to read from the jump vector table (JVT). Both instruction fetches are _implicit_ reads, and both require execute permission; read permission is irrelevant. `Only the first fetch is eligible for PC based debugging such as setting breakpoints.`

This is my proposed addition to the paragraph in the spec. Comments?

Silabs-ArjanB, 2022-05-16:
Looks good to me.

tovine, 2022-05-16:
What about for data watchpoint triggering? The suggested paragraph doesn't explicitly say that the JVT access can't trigger a load address trigger.

closed 2022-05-16 by tariqkurd-repo