riscvarchive/riscv-code-size-reduction #162: Name confusion with Zbb: cpop

tovine, 2022-05-03:
See https://github.com/riscv/riscv-bitmanip/issues/174

It seems that sometime late last year, the Bitmanip TG changed the name of the popcnt instruction to cpop - which in my opinion is dangerously close to `cm.pop` (although it was even worse before when it was `c.pop`).
Should we consider renaming this instruction, or just bet on it being different enough (and that most people won't write assembly anyway)?

tariqkurd-repo, 2022-05-03:
interesting observation!
I would say that it's not a problem as the assembler syntax doesn't usually include the `c.` suffix anyway so it doesn't need to include the `cm.` suffix either so I suspect they'll be `cpop` and `pop`. Also the parameter lists are dramatically different between the two so I don't think they'll be confused. `cpop {ra,s0-s2}, 32` would certainly return an error as it should be `cm.pop`

abukharmeh, 2022-05-07:
>  so it doesn't need to include the cm. suffix either

@tariqkurd-repo  LLVM objdump at the moment does show the cm prefix unlike the c prefix ! 
 
Regarding the assembler, it appears to expects and only work if "cm." is specified, unlike normal compressed instructions where it would work both ways, and auto compress instructions if in range. I reopened https://github.com/plctlab/llvm-project/issues/30 as I think this is a regression !