riscvarchive/riscv-code-size-reduction #163: Can jvt.base be declared as WARL?

Silabs-ArjanB, 2022-05-25:
The jvt.base bitfield is not declared as WARL (jvt.mode is declared as WARL). In order to avoid an adder when computing the table jump address for the following computation it would be preferred if it would be allowed to hardwire some of the jvt.base lsbs to 0:

> switch(XLEN) {
> 32: table_address[XLEN-1:0] = JVT.base + (index<<2);
> 64: table_address[XLEN-1:0] = JVT.base + (index<<3);
> }

For XLEN = 32 I would want to implement base[31:10] as actual registers and tie base[9:6] to 0 (which is not allowed in the current Zc specification)

tariqkurd-repo, 2022-05-26:
text added - I won't push v0.70.5 quite yet, I'll see if any more issues arise in the next week

closed 2022-05-26 by tariqkurd-repo

Silabs-ArjanB, 2022-05-27:
> Hi @Silabs-ArjanB - I think it makes sense for JVT.BASE to be WARL, following the example of MTVEC. MTVEC also says _the set of values the register may hold can vary by implementation_ so I have no problem adding the same text into the Zc spec (when I get time). Will that solve the problem?

Hi @tariqkurd-repo , yes that will solve the issue. It is indeed desired exactly for the same reason as why the address part of MTVEC is WARL.