riscvarchive/riscv-code-size-reduction #164: An illegal instruction error happened when doing executable test about cm.pop instruction.

yulong-plct, 2022-05-30:
I was recently doing executable tests with qemu emulator for zc extension. All subextensions had passed, but not zcmpe. Next, I will tell you what I found, and I hope to ask for help.

First, I want to verify the validity of the cm.pop instruction. I compiled the following code and confirmed that the instruction is generated.
code:
[image]
asm:
[image]

Then I executed that binary and got an **Illegal instruction** error.
So I opened the qemu emulator and gdb, trying to find the illegal instruction. I found out that it was not an illegal instruction, but an illegal address, as shown in the figure.
[image]
[image]

After jumping out of the function _fstat, the pc jumps to the address 0x40800148 for execution, and then reports an illegal instruction error.

We guess that the pop/popret address offset is asymmetrical, which leads to the jump to the wrong address, because  I find that the zcmp subextended address offset is symmetric. You can see the related content as shown in the picture.

use the objdump tool to generate asm files under the zcmp extension:
[image]

use the objdump tool to generate asm files under the zcmpe extension:
[image]

liweiwei90, 2022-06-10:
> Sorry for late reply, so basically RA register is not being restored correctly, for EABI version of push/ pop ?

Yes, The question is RA register is not being restored correctly. Following is the debug output in spike:

- before executing cm.push, ra is 0x0001983c

```
: reg 0  
zero: 0x00000000  ra: 0x0001983c  sp: 0x7ffff888  gp: 0x00021ac0
  tp: 0x00000000  t0: 0x000163cc  t1: 0x00016410  t2: 0x0001f390
  s0: 0x000212c0  s1: 0x00021614  a0: 0x00000001  a1: 0x7ffff8a0
  a2: 0x7ffff8a0  a3: 0x00002009  a4: 0x000212c0  a5: 0x00000001
  a6: 0x00000000  a7: 0x00000000  s2: 0x00000000  s3: 0x00000000
  s4: 0x00000000  s5: 0x00000000  s6: 0x00000000  s7: 0x00000000
  s8: 0x00000000  s9: 0x00000000 s10: 0x00000000 s11: 0x00000000
  t3: 0x00000000  t4: 0x00000000  t5: 0x00000000  t6: 0x00000000 
```

- execute cm.push, ra is pushed into [0x7ffff87c], sp is updated to 0x7ffff848

```
: run 1
core   0: 0x0001b728 (0x0000b86e) cm.push {ra, s0-s1}, -64
push 9 at 7ffff884, 21614
push 8 at 7ffff880, 212c0
push 1 at 7ffff87c, 1983c
: reg 0
zero: 0x00000000  ra: 0x0001983c  sp: 0x7ffff848  gp: 0x00021ac0
  tp: 0x00000000  t0: 0x000163cc  t1: 0x00016410  t2: 0x0001f390
  s0: 0x000212c0  s1: 0x00021614  a0: 0x00000001  a1: 0x7ffff8a0
  a2: 0x7ffff8a0  a3: 0x00002009  a4: 0x000212c0  a5: 0x00000001
  a6: 0x00000000  a7: 0x00000000  s2: 0x00000000  s3: 0x00000000
  s4: 0x00000000  s5: 0x00000000  s6: 0x00000000  s7: 0x00000000
  s8: 0x00000000  s9: 0x00000000 s10: 0x00000000 s11: 0x00000000
  t3: 0x00000000  t4: 0x00000000  t5: 0x00000000  t6: 0x00000000
```

- before executing cm.pop, sp is updated to 0x7ffff854(which is not resumed to the sp value after cm.push)
```

: until pc 0 1b74e
: reg 0
zero: 0x00000000  ra: 0x0001b74a  sp: 0x7ffff854  gp: 0x00021ac0
  tp: 0x00000000  t0: 0x00000005  t1: 0x00008807  t2: 0x00002190
  s0: 0x00000000  s1: 0x7ffff8a0  a0: 0x00000000  a1: 0x7ffff7f8
  a2: 0x00000000  a3: 0x00000000  a4: 0x62a301b9  a5: 0x00000000
  a6: 0x00000000  a7: 0x00000000  s2: 0x00000000  s3: 0x00000000
  s4: 0x00000000  s5: 0x00000000  s6: 0x00000000  s7: 0x00000000
  s8: 0x00000000  s9: 0x00000000 s10: 0x00000000 s11: 0x00000000
  t3: 0x00000000  t4: 0x00000000  t5: 0x00000000  t6: 0x00000000
```
- after executing cm.pop, ra is udpated to 0x21614(loaded from [0x7ffff888]) which is not the ra value pushed to the stack
```
: run 1
core   0: 0x0001b74e (0x0000be6e) cm.popret {ra, s0-s1}, 64
pop 9 at 7ffff890, 14e90
pop 8 at 7ffff88c, 7ffff8a0
pop 1 at 7ffff888, 21614
: reg 0
zero: 0x00000000  ra: 0x00021614  sp: 0x7ffff894  gp: 0x00021ac0
  tp: 0x00000000  t0: 0x00000005  t1: 0x00008807  t2: 0x00002190
  s0: 0x7ffff8a0  s1: 0x00014e90  a0: 0x00000000  a1: 0x7ffff7f8
  a2: 0x00000000  a3: 0x00000000  a4: 0x62a301b9  a5: 0x00000000
  a6: 0x00000000  a7: 0x00000000  s2: 0x00000000  s3: 0x00000000
  s4: 0x00000000  s5: 0x00000000  s6: 0x00000000  s7: 0x00000000
  s8: 0x00000000  s9: 0x00000000 s10: 0x00000000 s11: 0x00000000
  t3: 0x00000000  t4: 0x00000000  t5: 0x00000000  t6: 0x00000000
```

- we can see the value in [0x7ffff87c] is not changed: 

```
: mem 0 7ffff87c,
0x0001983c
```
After this, fetch segfault is triggered:
```
z  00000000 ra 00021614 sp 7ffff894 gp 00021ac0
tp 00000000 t0 00000005 t1 00008807 t2 00002190
s0 7ffff8a0 s1 00014e90 a0 00000000 a1 7ffff7f8
a2 00000000 a3 00000000 a4 62a301b9 a5 00000000
a6 00000000 a7 00000000 s2 00000000 s3 00000000
s4 00000000 s5 00000000 s6 00000000 s7 00000000
s8 00000000 s9 00000000 sA 00000000 sB 00000000
t3 00000000 t4 00000000 t5 00000000 t6 00000000
pc 00021614 va 00021614 insn       ffffffff sr 80006020
User fetch segfault @ 0x00021614
```

abukharmeh, 2022-06-24:
Reported as fixed !

closed 2022-06-24 by abukharmeh