riscvarchive/riscv-code-size-reduction #165: 16-bit instruction to shift by a register value

jalobaba, 2022-06-09:
I recently was reviewing the Zc spec and noticed it was missing 16-bit shift instructions by a register value. Zc only currently has 16-bit shift instructions that shift by an immediate value.

@tariqkurd told me these instructions haven't been thought of yet. Can we explore the code size benefit they provide and consider them for Zc?

Note that Arm Thumb-2 includes 16-bit shift by register instructions as do some other ISAs.

abukharmeh, 2022-06-14:
It should be fairly easy if we know what patterns we are trying to optimise, is this trying to achieve the same thing as #141 ?

tariqkurd-repo, 2022-06-14:
It's different, that one has an immediate shift distance and I think different source and destination registers. It's worth nothing that these wouldn't expand into an existing 3# bit encoding which might be a problem, bits it's certainly worth knowing how valuable they are

abukharmeh, 2022-06-14:
`It's worth nothing that these wouldn't expand into an existing 32 bit encoding` Why not SRL AND SLL ?  

[image]

This does not looks really promising unless I am missing something, I will let you know exact % savings later in the week, or perhaps Saturday.

jalobaba, 2022-06-14:
The proposed 16-bit shift instructions will expand into existing 32-bit instructions.

[image]

jalobaba, 2022-06-14:
Tariq, you asked if these were the instructions I was proposing:
     CM.SLL rsd', rsd', rs2'

Yes, exactly this with 2 registers and destructive shift.

As for out of range shifts (i.e. rs2' > 31), I'd match the behavior of the 32-bit RISC-V shift instructions. I just checked the spec and the 32-bit shift instructions just use the bottom 5-bit of rs2 so I would match this for the 16-bit versions and then there is no special behavior for out of range.

tariqkurd-repo, 2022-06-15:
yes - you're right - CM.SLL would be the 16-bit encoding of SLL, and similarly for SRL/CM.SRL, SRA/CM.SRA

give that there doesn't seem to be much value in these in our existing benchmark suit @jalobaba do you have an open source application which sees a benefit from them?

tariqkurd-repo, 2022-06-15:
> No, I don't have an open-source application that benefits from 16-bit
> shift by register value.
>
> We have several groups at Qualcomm with RISC-V code for their
> microcontroller subsystems. I'm starting to reach out to them and get an
> effort started to evaluate Zc on those code bases. Some use GCC and some
> use LLVM so we can use your preliminary versions of those to evaluate Zc.
> We can also try Ibrahim's script. Is that useful to you to help determine
> the benefits of 16-bit shift by register?
>
> What is the best way to analyze our internal code to determine the benefit
> of 16-bit shift by register? I can always grep through the disassembly but
> is there a better way? How did you determine that the benchmarks you are
> using to evaluate Zc don't have much benefit for 16-bit shift by register?
> I'm hoping I could use the same approach.

The script doesn't support that currently, but simplistically you can grep
for the 32-bit destructive forms and compare the saving from shrinking
those by 2 bytes with the size of the entire text section. That should be
simple enough.

jalobaba, 2022-06-15:
Thanks Ibrahim. I'll be working on this next week.

abukharmeh, 2022-06-15:
I would be happy if you submit a PR for it :) 

Even if it didn't up in the spec, it would help people try it on their applications etc. 

So what you need to do is 

1. Add a new action name, e.g `elif action == 'cm_srl'`
2. Use retrieve field function with required instructions list set as [srl], and additional condition as new lambda function that checks that the one of the source register and the destination register are equal, and all registers are within the compressed format (Can use utils.regwithin as that is just a function that checks if a register with an ABI name is from ones that fit the compressed encoding) 
3. Use apply_optimizations to patch the instructions, the most important factor is to change WoE  (Width of encoding to 16)

Then just use the script 

`hca -a cm_srl -f name_of_your_elf_fille`

abukharmeh, 2022-06-26:
Hi, as I promised earlier, here is a complete results from our benchmark set

[image]

I doubt we can justify the encoding space for these instructions with these results ! 

Are we happy to close this now @jalobaba @tariqkurd-repo

jnk0le, 2022-06-26:
0.09% and 0.05% for rsd`+rs2` instruction. Bang for the code point ratio seems to be in line with other instructions (`c.mul` is also 0.09%)

abukharmeh, 2022-06-26:
Yes, you are right, they would be 2^(3+3)=64 points instructions as C.MUL which has similar savings !

tariqkurd-repo, 2022-06-27:
HI @abukharmeh are we also considering CM.SRA? Are there any results for that one?
Also 0.7-0.8% total for some of the embench benchmarks isn't too bad.
I need to think about this more but don't have time today......

abukharmeh, 2022-06-27:
`what do people think?` I agree, I think its too late to add anything if we want to ratification this year, and I dont see the savings from these instructions big enough to justify any addition to the current revision of extension !

jnk0le, 2022-06-27:
>...otherwise we also risk failing to ratify anything this year... and that would certainly be worse for RISC-V in general than proceeding without them

>  think its too late to add anything if we want ratification this year,!

I see one significant issue with "adding zce insrcutions later"; fragmentation. And those are likely to end up not used by A profile packages at all

eg. Zcev1 goes into profile with mandatory vector, such profile is very likely to become defacto lowest common denominator for targeting software (and optionally RV64GC as "legacy" one). If Zcev2 brings extra 0.5% code size reduction - nobody would like to break with LCD as well as provide yet another binary for just that 0.5%.

zcm* situation is a bit better, but not entirely.

Silabs-ArjanB, 2022-06-28:
> what do people think? I agree, I think its too late to add anything if we want ratification this year, and I dont see the savings from these instructions big enough to justify any addition to the current revision of extension 

In our opinion it is definitely not worth putting ratification at risk by adding new instructions. Given that Zc* already consists of 7 groups and that likely/hopefully other ideas we will pop up, these instructions (and others) can be added later (if needed in a new sub-extension).

tariqkurd-repo, 2022-06-28:
yes exactly - and delaying Zc for several months for minimal extra code-size improvement will only make the fragmentation worse.

tariqkurd-repo, 2022-06-28:
agreed - so no change now - and we can consider these in the future.

jnk0le, 2022-06-28:
> consists of 7 groups
In current zce spec, i see only 4 (+1 for RVE) "fragmentation" groups, 3 (+1) of them being embedded only.

> yes exactly - and delaying Zc for several months for minimal extra code-size improvement will only make the fragmentation worse.

given that the current compiler results match exactly the script ones. Maybe the compiler proof step could be skipped for obviously simple things like sll/srl/sra in v0.80

tariqkurd-repo, 2022-06-29:
there will only be a v0.80 if the architecture review committee request it