riscvarchive/riscv-code-size-reduction #166: Relationship between address and register in push/pop instructions

Silabs-ArjanB, 2022-06-22:
The specification states for push/pop that 'the bytes may be loaded in any order.'

It is not completely clear what that really means. This issue relates to https://github.com/riscv/riscv-code-size-reduction/issues/130, but goes a bit deeper into trying to understand what above statement really means.

For example the following is stated:

> cm.push {ra, s0-s5}, -64
> 
> appears to software as:
> 
> sw s5, -4(sp)
> sw s4, -8(sp)
> sw s3,-12(sp)
> sw s2,-16(sp)
> sw s1,-20(sp)
> sw s0,-24(sp)
> sw ra,-28(sp)

It seems logical to me that 'the bytes may be loaded in any order' at least means that the following sequence for a cm.push {ra, s0-s5}, -64 is allowed as well:

> sw ra,-28(sp)
> sw s0,-24(sp)
> sw s1,-20(sp)
> sw s2,-16(sp)
> sw s3,-12(sp)
> sw s4, -8(sp)
> sw s5, -4(sp)

What is not clear to me if 'the bytes may be loaded in any order' also for example allows the following sequence (changing the relationship between address and register) for a cm.push {ra, s0-s5}, -64:

> sw ra, -4(sp)
> sw s0, -8(sp)
> sw s1,-12(sp)
> sw s2,-16(sp)
> sw s3,-20(sp)
> sw s4,-24(sp)
> sw s5,-28(sp)

Could you please comment if such an implementation is allowed?

The reason to want a 'sw ra, -4(sp)' as initial transaction is that it will improve the critical path in our core (https://github.com/openhwgroup/cv32e40x) as both the -4 and the 'ra' will not require further decoding of a push instruction (just the fact that it is a push is enough info needed to generate such a first transaction). Computing -28 and/or s5 in this example is much more involved timing wise. (Unfortunately for a pop there will not be a deterministic first transaction.)

tariqkurd-repo, 2022-06-23:
Hi @Silabs-ArjanB - interesting comment about the critical path.
The bytes in the block of memory can be accessed in any order - but they must be read from the correct locations.
You can reorder the stores (and loads for POP), but not change where the bytes go, so it's fine to access ra first or last in the sequence but it must be stored to -4(sp).

The point of the statement is that you can turn the whole block into a stream of bytes and issue them to memory as you wish as they are non-overlapping

tariqkurd-repo, 2022-06-24:
Hi @Silabs-ArjanB can I close this now, or do you have more concerns?

Silabs-ArjanB, 2022-06-24:
The answer is clear. Thank you. We can still likely fix the stated path by moving some logic from the predecoder in IF to the regular decoder in ID (it just might make the code a bit less clean)