riscvarchive/riscv-code-size-reduction #173: more flexible jvt

jnk0le, 2022-07-15:
@jhauser-us 
>The solution is to allocate table entries for CM.JT starting at
index 63 and growing downward, not upward from index 0.  It's my
understanding that currently the tools are being written to handle a
JVT with a specified address for the start of the table (entry 0) and
a maximum used index number (usually 64 or higher).  My suggestion is
for the tools to record three components for a JVT:  the minimum used
index number (0 to 64), the maximum used index number (64 to 255), and
the address of a known entry, which could be standardized as either
entry 0, the first used entry, or entry 64, whichever the tools find
most convenient.  If the minimum used index number M is > 0, the linker
should allow program code to be located in the space of table entries
0 to M - 1, so only the used table entries take up space.

putting code within highly aligned table is not the cleanest thing in the world.

I think it's possible to get rid of static allocation between `jt`/`jalt` so we can get better utilization of cost function, for specific binaries.

a) ra writeback controlled by lsb set in address entries (normlly ignored by `jalr`)

- ra writeback mux will be very late in the pipeline
- allows more cache friendly aggregation of jt/jalt

b) csr holding address/offset of last entry that (not-)writes to ra

- following entries will do the inverse
- replaces "minimum entry" csr in case of "gadget secured" arch targeting JH approach

aswaterman, 2022-07-17:
The table doesn't need to be highly aligned.  It can start at any multiple of 64 bytes, which isn't excessive in this context.  The table can be placed in the address space so that unused entries at the beginning of the table are in vacant address space, not wasting memory.  And of course the entries at the end can contain unrelated information, like rodata or code.  So, even when the number of entries of each type doesn't match the static allocation in the spec, there's little wasted space.

So, I don't think either option is necessary.

But I'll briefly mention that deciding late in the pipeline whether ra is written is not a trivial thing for some microarchitectures; conditional move instructions were omitted from the base ISA for similar reasons.  The problem is that deciding the microarchitectural source of dependent instructions' operands can be an early-binding decision.  (Renaming is a common reason, but it's not the only one.)  It's not that the problem isn't solvable; it's that it's a new microarchitectural pattern that we shouldn't dismiss as being straightforward to implement in general.

jnk0le, 2022-07-17:
So the first option is out.

There is still option for jalt/jt flexibility (like 248 `jalt` + 8 `jt`) at the cost of extra csr.

> The table doesn't need to be highly aligned. It can start at any multiple of 64 bytes, which isn't excessive in this context. The table can be placed in the address space so that unused entries at [...]

<details>
  <summary>Here is how gcc 9.3.1 (-Os) handles 4 byte holes from 8 byte aligned functions in T2:</summary>
  
ssame result with `-mcpu=cortex-m4` 

  ```
          __run_init_array:
0800097c:   ldr     r3, [pc, #56]   ; (0x80009b8 <__run_init_array+60>)
0800097e:   push    {r4, r5, r6, lr}
08000980:   ldr     r5, [pc, #56]   ; (0x80009bc <__run_init_array+64>)
08000982:   subs    r3, r3, r5
08000984:   cmp     r3, #0
08000986:   ble.n   0x8000998 <__run_init_array+28>
08000988:   asrs    r6, r3, #2
0800098a:   movs    r4, #0
0800098c:   adds    r4, #1
0800098e:   ldr.w   r3, [r5], #4
08000992:   blx     r3
08000994:   cmp     r6, r4
08000996:   bgt.n   0x800098c <__run_init_array+16>
08000998:   ldr     r3, [pc, #36]   ; (0x80009c0 <__run_init_array+68>)
0800099a:   ldr     r5, [pc, #40]   ; (0x80009c4 <__run_init_array+72>)
0800099c:   subs    r3, r3, r5
0800099e:   cmp     r3, #0
080009a0:   mov.w   r6, r3, asr #2
080009a4:   ble.n   0x80009b4 <__run_init_array+56>
080009a6:   movs    r4, #0
080009a8:   adds    r4, #1
080009aa:   ldr.w   r3, [r5], #4
080009ae:   blx     r3
080009b0:   cmp     r6, r4
080009b2:   bgt.n   0x80009a8 <__run_init_array+44>
080009b4:   pop     {r4, r5, r6, pc}
080009b6:   nop     
080009b8:   subs    r6, #228        ; 0xe4
080009ba:   lsrs    r0, r0, #32
080009bc:   subs    r6, #228        ; 0xe4
080009be:   lsrs    r0, r0, #32
080009c0:   subs    r6, #228        ; 0xe4
080009c2:   lsrs    r0, r0, #32
080009c4:   subs    r6, #228        ; 0xe4
080009c6:   lsrs    r0, r0, #32
          CM7_1T_AES_128_keyschedule_enc:
080009c8:   stmdb   sp!, {r4, r5, r6, r7, r8, lr}
080009cc:   ldmia.w r1, {r2, r3, r4, r5}
080009d0:   mov.w   r1, #16843009   ; 0x1010101
080009d4:   str.w   r2, [r0], #4
080009d8:   movw    lr, #1024       ; 0x400
080009dc:   str.w   r3, [r0], #4
080009e0:   movt    lr, #8192       ; 0x2000
080009e4:   str.w   r4, [r0], #4
080009e8:   and.w   r12, r5, #255   ; 0xff
080009ec:   eor.w   r2, r2, r1, lsr #24
080009f0:   mov.w   r8, r5, lsr #24
080009f4:   lsls    r1, r1, #1
080009f6:   uxtb.w  r7, r5, ror #16
080009fa:   ldrb.w  r12, [lr, r12, lsl #2]
080009fe:   uxtb.w  r6, r5, ror #8
08000a02:   ldrb.w  r8, [lr, r8, lsl #2]
08000a06:   bcc.n   0x8000a0c <CM7_1T_AES_128_keyschedule_enc+68>
08000a08:   mov.w   r1, #454761243  ; 0x1b1b1b1b
08000a0c:   ldrb.w  r7, [lr, r7, lsl #2]
08000a10:   eor.w   r2, r2, r12, lsl #24
08000a14:   ldrb.w  r6, [lr, r6, lsl #2]
08000a18:   eor.w   r2, r2, r8, lsl #16
08000a1c:   str.w   r5, [r0], #4
08000a20:   eor.w   r2, r2, r7, lsl #8
08000a24:   cmp.w   r1, #1819044972 ; 0x6c6c6c6c
08000a28:   eor.w   r2, r2, r6
08000a2c:   str.w   r2, [r0], #4
08000a30:   eor.w   r3, r3, r2
08000a34:   str.w   r3, [r0], #4
08000a38:   eor.w   r4, r4, r3
08000a3c:   str.w   r4, [r0], #4
08000a40:   eor.w   r5, r5, r4
08000a44:   bne.n   0x80009e8 <CM7_1T_AES_128_keyschedule_enc+32>
08000a46:   str     r5, [r0, #0]
08000a48:   ldmia.w sp!, {r4, r5, r6, r7, r8, pc}
08000a4c:   movs    r0, r0
08000a4e:   movs    r0, r0
          CM7_1T_AES_192_keyschedule_enc:
08000a50:   stmdb   sp!, {r4, r5, r6, r7, r8, r9, r10, lr}
08000a54:   movw    lr, #1024       ; 0x400
08000a58:   movt    lr, #8192       ; 0x2000
08000a5c:   ldmia.w r1!, {r2, r3, r4, r5, r6, r7}
08000a60:   mov.w   r1, #16777216   ; 0x1000000
08000a64:   stmia.w r0!, {r2, r3, r4, r5}
08000a68:   and.w   r12, r7, #255   ; 0xff
08000a6c:   mov.w   r10, r7, lsr #24
08000a70:   strd    r6, r7, [r0], #8
08000a74:   uxtb.w  r9, r7, ror #16
08000a78:   ldrb.w  r12, [lr, r12, lsl #2]
08000a7c:   uxtb.w  r8, r7, ror #8
08000a80:   ldrb.w  r10, [lr, r10, lsl #2]
08000a84:   eor.w   r2, r2, r1, lsr #24
08000a88:   ldrb.w  r9, [lr, r9, lsl #2]
08000a8c:   eor.w   r2, r2, r12, lsl #24
08000a90:   ldrb.w  r8, [lr, r8, lsl #2]
08000a94:   eor.w   r2, r2, r10, lsl #16
08000a98:   movs.w  r1, r1, lsl #1
08000a9c:   orr.w   r8, r8, r9, lsl #8
08000aa0:   eor.w   r3, r3, r2
08000aa4:   eor.w   r2, r2, r8
08000aa8:   eor.w   r3, r3, r8
08000aac:   eor.w   r4, r4, r3
08000ab0:   str.w   r2, [r0], #4
08000ab4:   eor.w   r5, r5, r4
08000ab8:   str.w   r3, [r0], #4
08000abc:   strd    r4, r5, [r0], #8
08000ac0:   bcs.n   0x8000ac8 <CM7_1T_AES_192_keyschedule_enc+120>
08000ac2:   eors    r6, r5
08000ac4:   eors    r7, r6
08000ac6:   b.n     0x8000a68 <CM7_1T_AES_192_keyschedule_enc+24>
08000ac8:   ldmia.w sp!, {r4, r5, r6, r7, r8, r9, r10, pc}
08000acc:   movs    r0, r0
08000ace:   movs    r0, r0
          CM7_1T_AES_256_keyschedule_enc:
  ```
  
</details>

It could move there 2 pc rel constants from previous function but it didn't

I wouldn't be so optimistic in more complex cases.

finding 64byte aligned spot with the least waste is easier (and already common) than like 64+44 that's at least 192 bytes away from 0x00000000

tariqkurd-repo, 2022-07-27:
Aligning the table differently to avoid memory wastage is a tools recommendation only, it doesn't change the semantics of the instructions.

I don't have time to review the spec at the moment but i'll leave this open to as a reminder to look at this again when I can.

jhauser-us, 2022-08-16:
The concerns expressed by Andrew become irrelevant if, rather than treating a JALT instruction as a form of combined load-and-branch, we instead assume a core reads from the jump vector table in the pipeline's "front end" as part of instruction fetch.  In this arrangement, JVT entries can be fetched and cached the same as ordinary instructions, and treated as oddly encoded JAL instructions.

Here's how it works in more detail:  On seeing a JALT instruction, the front end temporarily sets an internal JVT flag and jumps to the address of the table entry, which is then fetched the same as an ordinary instruction.  With the entry's bytes in hand, the JVT flag tells the core to treat them not as a regular instruction but rather a specially encoded JAL instruction.  (For RV64, a JVT entry is two 32-bit words, so two words must be fetched, probably in sequence, although that depends on the microarchitecture.)  Like any other JAL, the specially encoded JAL causes the front end to jump again to the address specified.  As it does, the front end remembers to clear the JVT flag, restoring normal instruction decoding.

The rest of the pipeline must also be aware of the specially encoded JAL instructions originating from JVT entries, so an extra bit or two must be kept for instructions progressing through the pipeline.  Instructions are thus expanded internally from 32 bits to 33 or 34 bits.  (There can be other reasons as well for widening instructions internally with a bit or two, completely independent of the JVT, so this widening may or may not be a new burden.)

Others may feel differently, but I think it is acceptable for Zcmt to be specified in a way that favors this "front-end" form of hardware implementation.  I spoke with Andrew and persuaded him that this was a reasonable assumption for Zcmt.

If we can assume that hardware implementations of JALT will normally be of the form described above, then I claim there should be no serious impediment to using bit 0 of JVT entries to indicate the linkage register:  0 = x0 (no linkage), 1 = x1.  My advice would be that this change be adopted, dispensing with the hard-coded break in the JVT between JT and JALT entries.  Responding to the original assertion at the top of this GitHub issue, I disagree that there would be any inherent difficulty in allocating JT and JALT entries outward from numbers 63 and 64.  But I do believe many people will find it more natural not to have to think about the table that way.

jalobaba, 2022-08-18:
I really like this idea. The added flexibility for compilers/linkers to allocate table entries freely between JALT and JT instructions instead of fixing JT to be the first 64 table entries could provide added compression. Did we measure this across the benchmark suite?

tariqkurd-repo, 2022-08-19:
I think this idea, which was the original proposal, is better as it offers more future flexibility in the contents of the table. In future we could put more stuff in there than just addresses, and it shows that the entry "is not just an address" it also has some control information in it.
So I think this future-proofs the table quite well, so I'm in favour.

jalobaba, 2022-08-30:
Can we accept John's proposal to use the LSB of the table entry to choose between jump no link and jump with link? I think it stands on its own merit but it also works better with some possible extensions to Zc in the future.

tariqkurd-repo, 2022-09-02:
We've gone with this proposal - please see the updated spec

closed 2022-09-02 by tariqkurd-repo

jnk0le, 2022-09-07:
upgrade of the second option:

Use the lowest 6 bits of `JVT.base` for specifying the last non-linking entry. Coarser granularity can give back one or two bits to keep them reserved or extend the applicable range above 64th entry.