riscvarchive/riscv-code-size-reduction #176: [EABI] RV32 only stack support instructions

jnk0le, 2022-07-24:
Due to 2x`XLEN` stack alignment, RV32 eabi needs extra compressed instructions to efficiently operate on 8 byte aligned stack, otherwise non compressed instructions will have to be used when limiting stack waste is important.

- push/pop with 8 byte stack granularity (probably 3rd imm bit?)- most obvious alternative is to not use push/pop, or use it for 3-4 reg chunks only and handle rest by `addi` + loads/stores

- `C(M).ADDI8SP` - probably not necessary, push/pop with 8 byte granularity can prepare initial 8 byte allocation for an following `addi16sp`. Non zcmp[e] EABI cores may need it.

tariqkurd-repo, 2022-07-27:
why 8-byte aligned? Is this in the EABI proposal?

jnk0le, 2022-07-27:
yes https://github.com/riscv-non-isa/riscv-eabi-spec/blob/master/EABI.adoc#4-eabi-stack-alignment

currently 2xXLEN for both RV32 and RV64