riscvarchive/riscv-code-size-reduction #177: [idea] x-register-indexed CSR access

stnolting, 2022-07-25:
I am not sure if this is the right place for something like this. Maybe [riscv/riscv-isa-manual](https://github.com/riscv/riscv-isa-manual) would the better place?! Anyway, this is just an idea I had. I am not sure about the "relevance" of this as this might be way too specific. However, it might still be interesting - if only for me! 😅

# Current State

The CSR access instructions from the `Zicsr` ISA extension use a **compile-time-static** 12-bit immediate to select one out of 4096 CSRs for being accessed. However, sometimes (often?) you see code like this that is used to "emulate" **random CSR access**:

```c
uint32_t csr_read(int csr_addr) {

  register uint32_t res = 0;

  switch (csr_addr & 0xfff) {
    // ...
    case 0x340: asm volatile ("csrr %[result], mscratch" : [result] "=r" (res) : ); break;
    case 0x341: asm volatile ("csrr %[result], mepc"     : [result] "=r" (res) : ); break;
    case 0x342: asm volatile ("csrr %[result], mcause"   : [result] "=r" (res) : ); break;
    case 0x343: asm volatile ("csrr %[result], mtval"    : [result] "=r" (res) : ); break;
    // ...
    default: res = 0; break;
  }

  return res;
}
```

(Another function of this type would be used to provide arbitrary CSR write access, another one for atomic CSR bit-set (`csrrs`) and another one for atomic CSR cit-clear (`csrrc`) - or all combined in one function with another "operation" specifier.)

This is especially handy for accessing contiguous CSR "sets" like the HPM counter registers or the PMP address/configurations registers. The drawback is that such "emulation functions" require additional code (getting quite big if you want to provide access to all 4096 CSRs) and also require several clock cycles to execute, which might be relevant for code related to **context changes**.

# Proposal

So how about adding new instructions for accessing a CSR with an **runtime-defined** address taken from a `x` register?

The current `Zicsr` instruction uses the I-type instruction format. For a register-indexed CSR instruction one could use the R-type format:

[image]

* `opcode` & `funct3`: see options below
* `rd`: destination register for CSR read data
* `rs1`: source operand for CSR to-be-written
* `rs2`: bits 11:0 defined the CSR address(`csr[ rs2[11:0] ]`); all remaining bits are ignored

So we would get something like this: read/write a CSR that is indexed by bits 11:0 of rs2.
```
csrrwx rd, [rs2], rs1  // rd <= csr[rs2[11:0]]; csr[rs2[11:0]] <= rs1
```

All "conditions determining whether a CSR instruction reads or writes the specified CSR" (RISC-V ISA manual, table 11-1) would also apply to this concept just like the CSR access permission checks (privilege level, CSR read-only, CSR implemented at all, etc.).

From a hardware point of view the additional logic required for such an instruction would be quite low (basically, just a 12-bit 2:1 mux).

I see two options how this kind of CSR access instructions could be added to the ISA:

## Option 1: "NEW" - Assign a new OPCODE

This would allow to implement all flavors of the `Zicsr` instructions - just with a register-based CSR address instead of the immediate-based one.

Pseudo-assembly using `csrr*x` to indicate CSR selection via rs2[11:0]:

```
csrrwx[i] rd, [rs2], rs1  // rd <= csr[rs2[11:0]]; csr[rs2[11:0]] <= rs1/uimm5
csrrsx[i] rd, [rs2], rs1  // rd <= csr[rs2[11:0]]; csr[rs2[11:0]] <= csr[rs2[11:0]] or rs1/uimm5
scrrcx[i] rd, [rs2], rs1  // rd <= csr[rs2[11:0]]; csr[rs2[11:0]] <= csr[rs2[11:0]] and not rs1/uimm5
```

Using a new opcode seems to be a "clean" approach as the instruction format (R-type) is defined entirely by the OPCODE.

## Option 2: "EXTEND" - OPCODE = SYSTEM; funct3 = 100

By using the so far undefined combination of OPCODE = SYSTEM and funct3 = `100` (not used in the current spec) one could
add a single "CSR read/write instruction with register-based CSR address". So no atomic set/clear options.

Pseudo-assembly using `csrrwx` to indicate CSR selection via rs2[11:0]:

```
csrrwx rd, [rs2], rs1  // rd <= csr[rs2[11:0]]; csr[rs2[11:0]] <= rs1
```

Here the instruction format (I-type (classic `Zicsr`) or R-type) is defined by the combination of OPCODE and funct3.

tariqkurd-repo, 2022-07-27:
can you send that to the unpriv mailing list? It's not a code-size-reduction proposal.

Thanks!

closed 2022-07-27 by tariqkurd-repo

stnolting, 2022-07-28:
> It's not a code-size-reduction proposal IMHO.

Haha, I thought so :wink:

> an you send that to the unpriv mailing list?

Sure! Could you point me somewhere? Or do you think ISA proposal could also be posted to [riscv/riscv-isa-manual](https://github.com/riscv/riscv-isa-manual)?