riscvarchive/riscv-code-size-reduction #178: update JVT.base to match the way privilege spec defines mtvec.base

dansmathers, 2022-08-22:
The spec currently shows 
{reg:[
    { bits:  6, name: 'mode',  attr: ['6'] },
    { bits: 26, name: 'base',  attr: ['XLEN-6'] },
],config:{bits:32}}

Can you change it to:
{reg:[
    { bits:  6, name: 'mode',  attr: ['6'] },
    { bits: 26, name: 'base[XLEN-1:6]',  attr: ['XLEN-6'] },
],config:{bits:32}}

This makes it explicit that base bits 5:0 are not in JVT register and matches how it is specified for mtvec.base in the privilege spec.   I had a question from someone wondering if JVT.base[5:0] refers to JVT[11:6] and I can see how they might think that based on the way it is currently in the spec.

tovine, 2022-08-25:
One more detail: `{bits:32}` implies that it's a 32-bit register (if my understanding is correct), but then we have `base[XLEN-1:6]` - implying that the bitfield is `XLEN-6` bits. What happens in the case where XLEN=64 (or potentially even larger)?

Would it make more sense to say `config:{bits:XLEN}`?

tariqkurd-repo, 2022-08-26:
yes - the fix for this is in progress but hasn't been committed yet - I've split the register definitions for RV32 and RV64

tariqkurd-repo, 2022-09-02:
because it' s URW register, the correct answer is UXLEN bits. Please see the updated spec.

closed 2022-09-02 by tariqkurd-repo