riscvarchive/riscv-code-size-reduction #179: `Zcmt` → `Zicsr` dependency?

a4lg, 2022-09-11:
As the `Zcmt` extension adds `jvt` CSR (URW, `0x17`), I think it's required to note that the `Zcmt` extension depends on `Zicsr` as well as `Zca`.

This is important also because the `jvt` CSR is an unprivileged CSR (so we cannot assume the existence of the privileged architecture, which requires `Zicsr`).

a4lg, 2022-09-11:
This is a duplicate of #155 (I didn't notice that when I opened this) but I could not find an actual fix for #155 (despite that this issues is closed as _fixed_).

tariqkurd-repo, 2022-09-12:
this is good point - I'm not sure why that dependency didn't come through. I'll leave this one open while I investigate.

a4lg, 2022-09-12:
For privileged specifications, it's okay.

For unprivileged specifications, the position is not *that* clear. To be clear, recent draft of the RISC-V ISA Manual, Part I (unprivileged part) is doing *the opposite*.  For instance, quoting [the latest manual (draft)](https://github.com/riscv/riscv-isa-manual/releases/tag/draft-20220910-418a1b2):

> ### Chapter 12 “Zicntr” and “Zihpm” Counters
> ...
> The `Zicntr` extension depends on the `Zicsr` extension.
> ...
> The `Zihpm` extension depends on the `Zicsr` extension.

I'm happy if `Zicsr` dependency is implicit as you said (because we can avoid existing problems like in `Zkr` (a part of cryptography extensions)) but... since the position is not clear, I request the changes (I'm not closing this yet).

tariqkurd-repo, 2022-09-12:
@aswaterman I'm sure this came up before and I didn't list Zicsr in the Zcmt spec for the reason listed above. Can you comment on @a4lg 's concerns?

a4lg, 2022-09-12:
@tariqkurd-repo Interesting. Yes, I need *some* proper clarification but implicit `Zicsr` dependency is okay for me (see below).

### Further Context

Similar discussion came up when I tried to implement implicit implication rule `Zkr` / `Zve32x` → `Zicsr` etc... to GNU Binutils and concluded that some clarification is needed somewhere.

* https://sourceware.org/pipermail/binutils/2022-September/122757.html (PATCH v1)
* https://sourceware.org/pipermail/binutils/2022-September/122761.html (PATCH v2)
* https://sourceware.org/pipermail/binutils/2022-September/122766.html (comment by @Nelson1225)
* https://sourceware.org/pipermail/binutils/2022-September/122768.html (comment by @kito-cheng)

At the time of PATCH v1 submission, I expected implied `Zicsr` rule as @tariqkurd-repo described. But further investigation made the situation unclear on unprivileged extensions and we had to temporally conclude that it's not safe to assume that implication rule on unprivileged extensions.

So, for me, both options are okay as long as clarified *somewhere*:

1.  There is an implicit dependency (`an unprivileged extension with CSR definitions/uses` → `Zicsr`)
2.  There is (or must be) an explicit dependency (likewise)

aswaterman, 2022-09-12:
Please just include the following sentence in the Zcmt spec: "The Zcmt extension depends on the Zicsr extension."

This is what we do for e.g. the F extension.  It means that listing only Zcmt in the ISA string is sufficient; Zicsr is implicitly present if Zcmt is present.

tariqkurd-repo, 2022-09-13:
so elsewhere we're using the phrase (e.g.) `Zcmt requires Zca`, as oppose to `Zcmt depends on Zicsr`, so the updated adoc text is:

`Zcmt requires the <<Zca>> and Zicsr extensions.`

do you take _requires_ and _depends_ on to have the same meaning here?

thisi is what I've put in v1.0.0-RC5.3 - but if the relationship between Zcmt and Zicsr and Zcmt and Zca are _different_ then I'd like to be clear about exactly what that means

Tariq

aswaterman, 2022-09-13:
I'm not trying to suggest the relationship between Zcmt and Zca differs from the relationship between Zcmt and Zicsr.  I'm just quoting the ratified spec.  We should use the same language for all of the above, IMO.

closed 2022-09-13 by tariqkurd-repo