riscvarchive/riscv-code-size-reduction #180: gcc: Zc toolchain does not seem to allow half-word aligned functions

silabs-kjetil, 2022-09-12:
When building code with the gcc Zc* toolchain supplied in this repo I notice that functions are always word aligned and padding bytes are added which makes the code size larger than expected. Is this a bug in the toolchain, or is there a way I can force the compiler to accept half-word aligned functions when building with the documented Zc* march argument

`-march=rv32ima_zca_zcb_zcmb_zcmp_zcmt -mabi=ilp32`

tariqkurd-repo, 2022-09-12:
Hi - I think you should put this issue here instead: https://github.com/plctlab/llvm-project

closed 2022-09-12 by tariqkurd-repo

abukharmeh, 2022-09-12:
Possible duplicate of https://github.com/plctlab/corev-gcc/issues/2
I thought this issue somehow was fixed in later versions, can you try forcing the alignment with `falign ` compiler options ?

silabs-kjetil, 2022-09-12:
Using `falign-functions=2` did not seem to fix the issue. @tariqkurd-repo the link you sent is for an llvm-project, however this was seen when using gcc.

Are the gcc submodule pointers in this repo up to date with the newest development of the zc extension in gcc?

abukharmeh, 2022-09-12:
Yes, that submodule points to the latest commit. What libc is it using, and how many variations you ended up after the compilation in install/riscv32-unkown-elf/lib ?

silabs-kjetil, 2022-09-12:
It's using newlib-nano as far as I know and we end up with one variation "rv32ima" in the riscv32-unknown-elf/lib/ folder. The multilib generator string is "rv32ima-ilp32--zca*zcb*zcmt*zcmb*zcmp", is this correct or is it missing something?

I see the compiler produce cm.push and cm.pop instructions, however it does not produce any table jump code. Is there anything specific I need to enable to get gcc to use table-jumps?

abukharmeh, 2022-09-12:
Yep, that didn't work with me, I had to specify the multilib generator string manually for it to generate the correct libs
`--with-multilib-generator="rv32ima_zca_zcb_zcmt_zcmb_zcmp-ilp32--;rv32ima_zca_zcb-ilp32--;rv32ima_zca_zcmt-ilp32--;rv32ima_zca_zcmb-ilp32--;rv32ima_zca_zcmp-ilp32--"
`
These are the various options I intended to evaluate !

abukharmeh, 2022-09-12:
Re table jump, that should work without anything specific, for me simply march=rv32imac_zca_zcmt resulted in the linker generating them !

silabs-kjetil, 2022-09-12:
Do you know why we specify `rv32ima_...` instead of `rv32imac_...`. I would think that you needed to enable the C extension as well as Zc* to get all the compressed instruction benefits.

abukharmeh, 2022-09-12:
No, Zca = C without Floats, you need to use that now instead of C since we reuse the some of the compressed floating point instructions encoding space.

silabs-kjetil, 2022-09-12:
I see, that makes sense.

When it comes to the table jump it seems like there is something inside libc which is triggering the inclusion/exclusion of the table jump functionality. When I use nano.specs which pulls in libc_nano.a then the table jump functionality is turned off. If I don't use nano.specs and pull in libc.a then table jump is present.

abukharmeh, 2022-09-12:
I am not sure what is causing that, two things come to my mind, either the linker search for setting up JVT CSR in the bootstrap code, or there is something differing in the linker script ?

silabs-kjetil, 2022-09-12:
When libc.a is added I see that a section called `.text.tbljal` is added however the table is empty and is not being filled with function entries. Maybe there is something missing in my linkerscript.

silabs-kjetil, 2022-09-15:
I added an issue to the corev-binutils-gdb project for the table jump issues https://github.com/openhwgroup/corev-binutils-gdb/issues/44.