riscvarchive/riscv-code-size-reduction #182: why

JoshCafe, 2022-09-19:
hi 

I found in GCC ra is placed in lowest address slot of GPR save area in stack frame. 
[image]
Same is obseved when -msave-restore is enabled. 

However, in zcmp ra is placed in highest address slot. 
[image]

The reserved storage order causes different cfi offset of GPRs saved in GC/msave-restore cases and zcmp case.
May i know why in zcmp ra is placed in highest address slot? and can we align it to GC/msave-restore cases?

BR, 
Josh

linsinan1995, 2022-09-20:
iirc, there is no regulation for the storage order (see [issue](https://github.com/riscv/riscv-code-size-reduction/issues/166)).

In the current GCC implementation, the storage order is supposed to be the same as save-restore (the placement order for push is ra, s0, s1, ..., s11) and have the same CFI order as save-restore, since this part in GCC was adapted from save-restore.

JoshCafe, 2022-09-20:
> iirc, there is no regulation for the storage order (see [issue](https://github.com/riscv/riscv-code-size-reduction/issues/166)).
>
> In the current GCC implementation, the storage order is supposed to [...]

Thanks for providing the previously discussed issue thread. (see https://github.com/riscv/riscv-code-size-reduction/issues/166).

From [tariqkurd-repo's reply](https://github.com/riscv/riscv-code-size-reduction/issues/166#issuecomment-1164081281), the bytes in the block of memory can be accessed in any order - but they must be read from the correct locations.

The toolchain you mentioned has no functional issue in running push and pop pairs, but has issues in debugging. Because the CFI generated from toolchain could not read from the correct location. 

Toolchain has to generate CFI offset info as per the zcmt spec, or the spec needs to reverse the push order. 

@tariqkurd-repo, could you please clarify here also, Thanks.

tariqkurd-repo, 2022-09-21:
what's the problem here? I don't know what CFI means.
Zcmt is frozen, and will soon go to public review, so it's very unlikely to change.

abukharmeh, 2022-09-22:
@JoshCafe I think you meant Zcmp not Zcmt and Call Frame Information (Part of DWARF) ?, I am not sure this matters because they are single instruction ?

abukharmeh, 2022-09-22:
It feels like this is needed for debugger to unwind stack frame, right  ? https://stackoverflow.com/questions/24462106/what-do-the-cfi-directives-mean-and-some-more-questions  I don't see any issue with the compiler generating different CFI information to match the implementation ?  IIRC the order was set like this to optimise for interpretability !

JoshCafe, 2022-09-23:
> It feels like this is needed for debugger to unwind stack frame, right ? https://stackoverflow.com/questions/24462106/what-do-the-cfi-directives-mean-and-some-more-questions I don't see any issue with the compiler generating different CFI information to match the implementation ? IIRC the order was set like this to optimise for interruptibility!

> @JoshCafe I think you meant Zcmp not Zcmt and Call Frame Information (Part of DWARF) ?, I am not sure how this matters when they are single instructions ?

yes. I mean Zcmp and Call Frame Information (Part of DWARF) which is needed for debugger to unwind stack frame.

https://sourceware.org/binutils/docs/as/CFI-directives.html has a breif description of the CFI directives.

If user compiles with -g compiler option, CFI-directives  will be generated.  As cm.push grows the stack and store GPRs into it, .cfi_adjust_cfa_offset offset and multiple .cfi_offset register, offset will be generated accordingly.

let me reuse example from @Silabs-ArjanB(see https://github.com/riscv/riscv-code-size-reduction/issues/166)

This is the order from Zcmp spec:
sw s5, -4(sp)
sw s4, -8(sp)
sw s3,-12(sp)
sw s2,-16(sp)
sw s1,-20(sp)
sw s0,-24(sp)
sw ra,-28(sp)

This is the order [current toolchain](https://github.com/plctlab/corev-gcc) implements. It alligns with what GCC does for save-restore from @linsinan1995 .
sw ra, -4(sp)
sw s0, -8(sp)
sw s1,-12(sp)
sw s2,-16(sp)
sw s3,-20(sp)
sw s4,-24(sp)
sw s5,-28(sp)

If a crash happens between cm.push and cm.popret, the CFI info will be used to unwind stack frame. In this case, offset -4, as in sw ra, -4(sp), will be used as the stack frame offet to take ra, while in hardware, as per Zcmp spec,  offset -4 is actually the slot for s5. So this match will cause an error in unwinding the stack frame. 

As @tariqkurd-repo mentioned, Zcmp spec if frozen, we has to take this issue into consideration in toolchain.

tariqkurd-repo, 2022-09-23:
Hi @JoshCafe , you're right, the toolchain will have to be updated to support this.

closed 2022-09-23 by tariqkurd-repo

tovine, 2022-09-23:
I don't remember what the justification was (if any) for the current ordering, but what would be the point against aligning with the ordering from save-restore in the toolchain (obviously apart from the fact that it's frozen)?
Guess it depends on what's easier to change - spec or toolchain, and if all toolchains do it the same way or not