riscvarchive/riscv-code-size-reduction #183: error when run `sh gcc.sh`

lhtin, 2022-09-19:
When I try to compile the toolchain, I get the below error message. I think the repo currently doesn't include `utils/prepare-libgmp.sh`. Am I missing any information? Thanks.

```
➜  toolchain git:(main) sh gcc.sh
/usr/local/bin/gcc
/usr/local/bin/g++
gcc.sh: line 35: utils/prepare-libgmp.sh: No such file or directory
```

abukharmeh, 2022-09-19:
Hi
You can obtain that file from here for now https://raw.githubusercontent.com/embecosm/embecosm-toolchain-releases/master/utils/prepare-libgmp.sh , Also please note that multilib generator string needs fixing as it appears that the wild cards are not working, not sure why ! 
`--with-multilib-generator="rv32ima_zca_zcb_zcmt_zcmb_zcmp-ilp32--;rv32ima_zca_zcb-ilp32--;rv32ima_zca_zcmt-ilp32--;rv32ima_zca_zcmb-ilp32--;rv32ima_zca_zcmp-ilp32--"`
Thanks,
Ibrahim

lhtin, 2022-09-19:
Thank you for the comment.  Why not include this file in this repo instead obtain from extra?

abukharmeh, 2022-09-19:
Yes we should and will at some point !

lhtin, 2022-09-20:
Get it, thank you.

closed 2022-09-20 by lhtin