riscvarchive/riscv-code-size-reduction #185: Reduce  "long" 64 bit calls using AUIPC /JALR  to 48 bit.

RogierBrussee, 2022-09-24:
Calls  beyond the ±1Mbyte range (but within ±2Gbyte) use the sequence of two 32 bit instructions

AUIPC ra imm20
JALR  ra ra imm12
 
The processor is allowed to fuse the two immediates to one long immediate (SEXT(imm20)<<12 + SEXT(imm12))&~0x1 so the last bit of the imm12 is irrelevant and in practice always 0, i.e. imm12 == (imm11 <<1).  Effectively the sequence defines 64bit JAL instruction with an (essentially) 31 bit immediate.

There seem to be 2 ways to do reduce a "long call" from 64 to 48 bit: 

1. define a proper 48 bit instruction Z48bitinstr.JAL rs1 imm31 which is like JAL but with a 31 bit immediate (with perhaps rs1 reduced to 1 bit to only allow ra and zero)
2. define  a 16 bit instruction CM.JALRA imm11 which expands to JALR ra ra (imm11 <<1)  that is supposed to be used by the assembler to match and replace existing JALR  ra ra (imm11<<1) sequences. Hence the above 2*32 bit instruction sequence gets
replaced with the 32 + 16 bit instruction sequence

AUIPC ra imm20
CM.JALRA imm11

The first approach is the most direct, but requires the introduction of 48 bit instructions, for which there seems to be considerable hesitation,  but allows to make long jumps and hence tail calls. 
The second approach requires an entire opcode in the compressed code space and does not allow to compress 64bit tail calls but fits in with the current toolchains and, I would assume, instruction decoding pipelines.

Of course the table based approach of cm.jalt may reduce the number of 64 bit calls significantly, but this reduction is somewhat orthogonal and in some sense more conventional.

tariqkurd-repo, 2022-09-26:
Hi @RogierBrussee , cm.jalt/cm.jt are specifically designed to replace these AUIPC+JALR calls with a 16-bit encoding.
they also have the additional benefit of accessing the entire XLEN width address space, not limited to ±2Gbyte on RV64, they can access everywhere.

So what's the advantage of inventing something new?

Tariq

jalobaba, 2022-09-26:
Rogier, what problem are you trying to solve that isn't already solved by the table jump instructions? Are you trying to come up with a solution that offers compression without using table jump? If so, your compression isn't competitive with table jump since you are just reducing 64 bits to 48 bits whereas table jump reduces 64 bits to 16 bits. Table jump is limited to 256 entries but that it plenty large enough to encode the frequent jumps in an embedded program (RVM profile). It is easier to add table jump to a microcontroller than support 48-bit instructions. I think 48 bit instructions were explored earlier in the Code Size working group and were rejected already due to the design/verification impact on cores.

If you are worried about the performance impact of table jump, I wouldn't solve it by using more "conventional" instructions like you propose. Instead, I'd add a BTB-like structure to a microcontroller to cache the most frequently executed table jump instructions which would reduce the penalty of reading the table from memory (usually one cycle for a microcontroller). The BTB is checked in parallel with an instruction fetch and if it hits in the BTB, the instruction fetch is ignored and the BTB contents are used which contain the target jump address (could even provided the target instruction contents to create zero-delay branches that hit in the BTB).

RogierBrussee, 2022-09-26:
The advantage is that the table based approach can deal with only a  limited number of functions and requires a memory access (although I can see that one could use a cache for the physical address of the subroutine keyed off the index rather than the virtual address).  Software with more than 256 functions (an OS Kernel kuch) would, I guess, use the table for the most commonly called functions but any call outside of the table would still need to use JAL or AUIPC + JALR.  

In particular all (larger)  systems that cannot/don't want to use the Zcmt extension would benefit. From this point of view it is a pity that the last remaining slot in the C extension is now taken up for C.L/SB[U] and C.L/SH[U], as JALR ra ra calls should be common in larger general purpose systems and they would benefit from a c.JALRA instruction in a straightforward manner,  although I guess in particular C.LBU and C.SB have their benefits for such systems too.

tariqkurd-repo, 2022-09-26:
analysis of commonly called functions shows that very few contribute to most of the code size saving, so in the code we benchmarks 256 entries was more than enough.

tariqkurd-repo, 2022-09-26:
I'l lclose this, if there's more to say @RogierBrussee then please reopen it

closed 2022-09-26 by tariqkurd-repo