riscvarchive/riscv-code-size-reduction #186: Is there a gcc support mzce-cdecbnez/decbnez

fanghuaqi, 2022-09-26:
Hi there, 

    I found there are some experiment instructions in https://github.com/riscv/riscv-code-size-reduction/blob/main/Zc-specification/changes_since_v0.50.adoc#instructions-in-v050-but-not-in-v070, is there any riscv gcc support in it.

Thanks
Huaqi

linsinan1995, 2022-09-26:
There is one v0.50 gcc toolchain implementation from PLCT.

You can try decbnez there. However, these repos have not been updated for months, and I think there is no plan to maintain v0.50 implementation.

binutils:
https://github.com/linsinan1995/riscv-binutils-gdb/tree/riscv-binutils-experiment-zcea
gcc:
https://github.com/linsinan1995/riscv-gcc/tree/riscv-gcc-experiment-zcea
glibc:
https://github.com/linsinan1995/riscv-glibc/tree/newlib-4.1.0

fanghuaqi, 2022-09-26:
Ok, thank you @linsinan1995

closed 2022-09-26 by tariqkurd-repo