riscvarchive/riscv-code-size-reduction #187: Description of state enable

Silabs-ArjanB, 2022-09-26:
Could the description of State Enable in section 3.3 JVT CSR be changed as follows?

Current text:

> If bit 2 of a controlling stateen0 CSR is zero, then access to the JVT CSR and
> execution of a cm.jalt instruction result in an Illegal Instruction trap (or, if appropriate, a Virtual
> Instruction trap).

Proposed text:

> If bit 2 of a controlling stateen0 CSR is zero, then **lower privilege** access to the JVT CSR and
> **lower privilege** execution of a **cm.jt or** cm.jalt instruction result in an Illegal Instruction trap (or, if appropriate, a Virtual
> Instruction trap).

(to make it clear that the state enable CSRs only affect the behavior in lower privilege modes and to include cm.jt in the description).

tariqkurd-repo, 2022-09-26:
yes - cm.jt was missing
and yes I think you're right, I'll update it

tariqkurd-repo, 2022-09-26:
I'll roll this into the next release (I'm sure more problems will be found). Thanks!

closed 2022-09-26 by tariqkurd-repo