riscvarchive/riscv-code-size-reduction #191: Document psABI changes for Zcmt

abukharmeh, 2022-10-21:
For the spec ratification, we need to document the following 3 items 

1. Document the section name for the table. (e.g. .riscv.tbljalvec)
2. Document the symbol name of the section (e.g. __tbljalvec_base$)
3. Document how the linker optimizes (or relaxes) the program by zcmt.

Item 2 and 3 could be documented in a similar fashion to the `Global-pointer Relaxation`  while item 1 needs to be documented in special-section document

abukharmeh, 2022-10-21:
@linsinan1995 @Xinlong-Wu  Could you please have a go at documenting what is there or what we think is needed and create a PR to the psABI repo
If someone else think they are able to do this documentation work, it would be really helpful

abukharmeh, 2022-10-21:
This presentation gives details about the relaxation process, though I am not sure that's the ideal way to do it ! https://docs.google.com/presentation/d/1ilfCIc9Bv4Ryl2Uyl3uwYtANlh9GSIroQC8Z-VIfUPk/edit#slide=id.g10c480b750a_0_274

linsinan1995, 2022-10-21:
I can help document the detail, but I would suggest we first decide and document the section and symbol name in ZC spec.

Xinlong-Wu, 2022-10-24:
sure, I'm glad I could be of help

abukharmeh, 2022-10-25:
Initial [PR](https://github.com/riscv-non-isa/riscv-elf-psabi-doc/pull/349) was created by @linsinan1995 thank you very much
@Xinlong-Wu  Please review it and check if there is anything that need to be added !

tariqkurd-repo, 2022-10-28:
is this confirmed? Should I add 

`Document the section name for the table. (e.g. .riscv.tbljalvec)`
`Document the symbol name of the section (e.g. __tbljalvec_base$)`

to the spec?

abukharmeh, 2023-01-10:
I think we can close this one now, do we need to document section name and symbol name in Zc* document, or psABI documentation would suffice ?

closed 2023-01-10 by abukharmeh