riscvarchive/riscv-code-size-reduction #192: is it possible to add a section to the spec that summarizes and names the new ISA formats?

dansmathers, 2022-10-26:
The riscv-spec lists R-type, I-type, S-type, U-type and has a fibure that lists the RISC-V base instruction formats.   Can something similar be added to the ZC spec to be consistent?

e.g.
// ---------------------------------------------------------
// |15    13|   12  |11    10|9   8    7|6     5|4  3 2|1  0|
// ---------------------------------------------------------
// | funct3 |funct12| imm    | rs1'     | imm   |  rd' | op | CMB cm.lbu(C2), cm.lhu(C2), cm.lb(C0), cm.lh(C0), cm.sb(C0), cm.sh(C0)
// | funct3 |   funct12_10   | rs1'     | imm   |  rd' | op | CB1 c.lbu(C0)
// | funct3 |   funct12_10   | rs1'     |f6 |imm|  rd' | op | CB2 c.lhu(C0), c.lh(C0), c.sh(C0)
// | funct3 |   funct12_10   | rs1'     | imm   |  rs2'| op | CB3 c.sb(C0) 
// | funct3 |   funct12_10   | rsd'     | funct2|  rs2'| op | CB4 c.mul(C1)
// | funct3 |   funct12_10   | rsd'     | funct2|func42| op | CB5 c.zext.b(C1),c.sext.b(C1),c.zext.h(C1),c.sext.h(C1),c.zext.w(RV64,C1),c.not(C1),
// | funct3 |   funct12_8          |   rlist       |imm| op | CPE cm.push(C2), cm.pop(C2), cm.popretz(C2), cm.popret(C2)
// | funct3 |   funct12_8          |   index           | op | CMT1 cm.jt(C2)
// | funct3 |   funct12_10   |         index           | op | CMT2 cm.jalt(C2)
// | funct3 |   funct12_10   | sreg1    | funct2| sreg2| op | CP cm.mvsa01(C2), cm.mva01s(C2)
// ---------------------------------------------------------

tariqkurd-repo, 2022-10-28:
I don't believe these formats currently exist for 16-bit encodings.....

dansmathers, 2022-10-28:
right.  I think these are all new formats that Zc has created, right?

tariqkurd-repo, 2022-10-28:
probably - I haven't checked them - but my point is that all 16-bit encodings should have predefined formats - so I think this should be an issue against the RISC-V spec to add formats for all of them. There's no point creating formats for a subset.

dansmathers, 2022-10-28:
The C extension created predefined formats for a subset:
Format Meaning 15 14 13 12 11 10 9 8 7 6 5 4 3 2 1 0
CR Register funct4 rd/rs1 rs2 op
CI Immediate funct3 imm rd/rs1 imm op
CSS Stack-relative Store funct3 imm rs2 op
CIW Wide Immediate funct3 imm rd Œ op
CL Load funct3 imm rs1 Œ imm rd Œ op
CS Store funct3 imm rs1 Œ imm rs2 Œ op
CA Arithmetic funct6 rd Œ/rs1 Œ funct2 rs2 Œ op
CB Branch funct3 offset rs1 Œ offset op
CJ Jump funct3 jump target op
Table 16.1: Compressed 16-bit RVC instruction formats.

Shouldn't the Zc extension be consistent with that?  Or is the Zc extension compatible with table 16.1?  Based on the decodings I thought Zc created new formats.

tariqkurd-repo, 2022-10-28:
ok - I hadn't spotted that section. @aswaterman I guess we need to add new formats for the new encodings? Any idea of naming etc?

aswaterman, 2022-10-28:
Can you propose something that feels consistent with the spirit of the existing names, and I’ll provide feedback?

I would not consider all of the ones Dan mentioned to be distinct formats, though. If the register access patterns are the same, we don’t need to go out of our way to call them new formats and give them new names. Seems like busy work.

jalobaba, 2022-11-01:
Tariq and I have agreed that I'll take a crack at this.

jalobaba, 2022-11-01:
Dan, where did you come up with your list of instruction formats for the Zc extension? I don't see such a list in the Zc spec or anywhere in the code size TG GitHub. Did you make the list yourself by looking through each instruction in the Zc spec?

dansmathers, 2022-11-01:
I just made it myself when I was doing my implementation.

jalobaba, 2022-11-03:
Okay, here's my first crack at it. I took what Dan provided and removed the instructions that have been dropped from Zc. I took Andrew's suggestion to not create a new format if there are just register usage differences between instructions. This is how it was done in the C extension so there is precedent. Before we start discussing the proposed format names I came up, how does the choice of formats look to others (i.e., let's first agree on the required formats and then discuss names after that).

[image]

jalobaba, 2022-11-04:
Any feedback on my proposed Zc formats?

tariqkurd-repo, 2022-11-07:
I think they look reasonable (sorry for the slow response). Why is one of them `CA` without a `Z` prefix?
In fact @aswaterman what do you think of the naming scheme? There will be future 16-bit encodings so do we need a formalised way of allocating names?

jalobaba, 2022-11-08:
The format without a Z prefix ("CA") is named as such because it matches an existing C extension format except the register roles (e.g., src vs dst) are somewhat different. The MOV2 instructions in Zc use the 2 3-bit register fields differently (either both sources or both destinations). 

I can imagine arguments for reusing the existing C format vs. defining a new format. I chose the former since the there are many other C and Zc formats that are the same except for registers roles.

jalobaba, 2022-11-08:
Andrew, by fully render, do you mean modify the ADOC source for the Zc spec and adding the formats? If so, I'm not up-to-speed on ADOC and started looking at it but realized there is a pretty big learning curve and probably not reasonable for me to learn all that for just some minor spec changes. Tariq or Ibrahim, can you make the required ADOC changes and then get feedback?

aswaterman, 2022-11-08:
@jalobaba Yeah, it’s easier to determine visually, and we obviously need to code it up at some point. I’m assuming someone else in the code size TG can take this on if you aren’t yet up to speed.

tariqkurd-repo, 2022-11-08:
I don't have time for this at the moment. Once we have settled on the spec update related to https://github.com/riscv/riscv-code-size-reduction/issues/195 I'll make a pre-release spec update and circulate it for comment, i.e. do it all in one go (including any other feedback which needs fixing)

jalobaba, 2022-11-08:
Good point Dan. The current C extension has a naming scheme "functN" where N is the width of the field in bits. So, I guess the Zc f6 would be renamed funct1 and func4_2 would be renamed funct3 but there already is a funtc3. I don't think matching the C extension naming scheme (i.e. functN where N is width) is scalable for Zc and future needs.

Dan I see you used a naming convention of "fX" where X is the bit position for a 1-bit field and "functXY" where X is the MSB and Y is the LSB for a multi-bit field.

How about we use "fX" for a single bit field in bit position X and "fX_Y" for a multi-bit field from MSB X to LSB Y? So, then f6 would stay f6 and func4_2 would be f4_2.

tariqkurd-repo, 2022-11-21:
@dansmathers @jalobaba are you going to continue with this proposal?

dansmathers, 2022-11-21:
it looks fine to me.  I think we are waiting on others to review, right?

tariqkurd-repo, 2022-12-30:
I've finally found some time to look at this during my christmas break.

please see this: https://github.com/riscv/riscv-code-size-reduction/pull/204

I think that functN etc. doesnt need to be unique within all encodings - it's just an N-bit opcode, as in the existing formats I see functN values reused in different encoding fields.

I've separated load and stores as the register mapping is different, and I've seperated c.mv* as the s1/s2 register format is different.

I've put @aswaterman and @abukharmeh on review - I couldn't add @jalobaba or @dansmathers presumably as they dont ahve access rights to this repo

let me know what you think

Tariq

abukharmeh, 2023-01-10:
I think we can close this one as #204 was merged

closed 2023-01-10 by abukharmeh

tovine, 2023-02-10:
Semi-related: I've made a small fix to the latex table generation in the riscv-opcodes repo: https://github.com/riscv/riscv-opcodes/pull/160

Would be useful to have an instruction listing in the same style as the RV64G table in the unpriv. ISA manual, so I made this script a bit more flexible.
Not sure what direction the ISA manual will take in the future wrt. LaTeX vs. AsciiDoc (and this is maybe a small sidetrack from this issue) but I think it would be nice to have a table at the end of each extension chapter that summarizes all the encodings from that extension in a similar way.