riscvarchive/riscv-code-size-reduction #196: zero load in popretz doesn't need to be atomic

jnk0le, 2022-11-13:
Previous content of `a0` doesn't matter and it's no longer moving from `s` registers. I think it can be moved into non atomic part.

tariqkurd-repo, 2022-11-18:
You can do this in your implementation, but I don't see any need to update the spec

closed 2022-11-18 by tariqkurd-repo

jnk0le, 2022-11-19:
in 2.9 there is a note saying :

> The li a0, 0 could be executed more than once, but is included in the atomic section

meanwhile in 2.1 we have statements like:

>The optional li a0, 0, stack pointer adjustment and optional ret must only be committed only
when it is certain that the entire POP/POPRET instruction will commit.

> these must only execute once, will only execute after all loads
> complete successfully all instructions must execute atomically
> therefore these updates are not visible in the interrupt/exception handler
> li a0, 0

What's the purpose of all of those "must" words (and atomic inclusion) related to zero load if it can be implemented on the implementors will?

Going further. If implementors decide to do zero load in a non atomic way, will it be actually Zcmp compliant still?

I think that fixing this inconsistency is a backwards compatible change and could be easily handled by puplic review process.

jnk0le, 2022-11-19:
also the:
>The li a0, 0 could be executed more than once, but is included in the atomic section

could be interpreted as if it was allowed to be executed multiple times but not allowed to have it's result visible/retired after `popret` being interrupted before the atomic section.