riscvarchive/riscv-code-size-reduction #198: Is there a short name for Zc* extension

fanghuaqi, 2022-11-18:
Hi @tariqkurd-repo 

      When I want to declare I want to use all the features of Zc* extension, currently I need to specify `zca_zcb_zcmb_zcmp_zcmt`, this name is too long to remember and use, can you declare a simple short name which stands for require all the Zc* features, I think IP vendors would implement a full featured Zc extension, and want to let user easily use it in toolchain with shorter name.

Thanks
Huaqi

tariqkurd-repo, 2022-11-18:
It's a good question. @aswaterman @kasanovic is there a plan for a shorter name for the whole set?

aswaterman, 2022-11-18:
No.  The plan is to use profile names to shorten these strings, e.g., RVA23 will imply C and Zcb, and a future embedded profile might imply a broader set of these extensions.

fanghuaqi, 2022-11-18:
But it will cause normal user hard to use the `-march` option, why not just use `Zcall` or `Zc` to stand for full set of `Zc` extension.

aswaterman, 2022-11-18:
Something like `Zcall` is inherently ambiguous, because some `Zc*` extensions are incompatible with each other.

Please be patient and wait for profiles to solve this problem.

fanghuaqi, 2022-11-18:
Maybe gcc compiler could help to handle the incompatible when using `Zcall`.

aswaterman, 2022-11-18:
No, GCC should not start defining its own RISC-V extension names.

Profiles will solve this problem.  Please patiently wait for them to arrive.

fanghuaqi, 2022-11-21:
Profiles can help to solve part of the long isa name issue, if rv cpu not following profile requirement would have to use the long isa name in march option.

So I thought for each extension, if it can define a full feature set isa name, it would be good for user to remember and use it.

tovine, 2022-11-21:
I see your point and understand where the desire for a short name is coming from, but I don't think it's a big problem in most cases.
Usually you have some sort of build system managing compiler options for you (e.g. make, CMake, ninja, or at the minimum some shell script) so you really only have to write the march string once per project.

fanghuaqi, 2022-11-21:
but for IP vendors, it need to provide configurable RISC-V IP, so the march string will need to be easily handled by user, user can easily turn on or off selected extension.

tariqkurd-repo, 2022-11-21:
"please wait for the profile" is the conclusion