riscvarchive/riscv-code-size-reduction #200: Consider move targets for cm.jalt to the front of the table

linsinan1995, 2022-11-23:
An experiment was done on embench-iot based on spec 0.70, and it shows that if we move the entries for cm.jalt to the front of the table, we could gain around 100 bytes of size saving for each case on average. 

```
+----------------+----------+-------------------+-------------+
|      case      | original | jalt at the front | size change |
+----------------+----------+-------------------+-------------+
| aha-mont64     |    16630 |             16534 |         -96 |
| crc32          |    15774 |             15680 |         -94 |
| cubic          |    41882 |             41762 |        -120 |
| edn            |    17096 |             17010 |         -86 |
| huffbench      |    17368 |             17264 |        -104 |
| matmult-int    |    16180 |             16094 |         -86 |
| md5sum         |    56542 |             56528 |         -14 |
| minver         |    16624 |             16538 |         -86 |
| nbody          |    16312 |             16204 |        -108 |
| nettle-aes     |    18302 |             18206 |         -96 |
| nettle-sha256  |    21162 |             21070 |         -92 |
| nsichneu       |    31020 |             30922 |         -98 |
| picojpeg       |    23472 |             23332 |        -140 |
| primecount     |    15830 |             15736 |         -94 |
| qrduino        |    21550 |             21452 |         -98 |
| sglib-combined |    20338 |             20272 |         -66 |
| slre           |    18348 |             18246 |        -102 |
| st             |    16410 |             16300 |        -110 |
| statemate      |    19274 |             19168 |        -106 |
| tarfind        |    15988 |             15894 |         -94 |
| ud             |    16372 |             16278 |         -94 |
| wikisort       |    19808 |             19684 |        -124 |
+----------------+----------+-------------------+-------------+
average saving: 95.82
```

The idea is simple. The table is prunable and function calls are more common cases to gain size saving in zcmt. Take aha-mont64(in rv32) as an example,

```
# N.B. each entry takes 4 bytes of space in the table. Thus, an entry saving 4 bytes can
# bring no extra size saving and a direct jump is converted into a table jump, which is bad
# for performance. 

cm.jt:
	index=0, sym name=.L45, address=0x00010696, savings=10
	index=1, sym name=_fwalk_reent, address=0x0001218c, savings=6
	index=2, sym name=__malloc_unlock, address=0x000134ae, savings=4
	index=3, sym name=.L175, address=0x00010d7a, savings=4
	index=4, sym name=.L110, address=0x00010f84, savings=4
	index=5, sym name=_free_r, address=0x000122c6, savings=4
	index=6, sym name=_getpid, address=0x00014638, savings=2
	index=7, sym name=.L167, address=0x00010d34, savings=2
	index=8, sym name=.L221, address=0x00010d2e, savings=2
	index=9, sym name=.L203, address=0x00010eea, savings=2
	index=10, sym name=__sbprintf, address=0x000118f6, savings=2
	index=11, sym name=.L157, address=0x00010d12, savings=2
	index=12, sym name=.L172, address=0x00010d72, savings=2
	index=13, sym name=.L24, address=0x00010c78, savings=2
	index=14, sym name=.L42, address=0x00010c6c, savings=2
	index=15, sym name=_malloc_r, address=0x00012bd0, savings=2
	index=16, sym name=.L40, address=0x00010bc8, savings=2
	index=17, sym name=_fwalk, address=0x00012860, savings=2
	index=18, sym name=_vfiprintf_r, address=0x00010ace, savings=2
	index=19, sym name=.L86, address=0x00010e40, savings=2
	index=20, sym name=atexit, address=0x000141b6, savings=2
	index=21, sym name=.L214, address=0x00010f72, savings=2
	index=22, sym name=.L84, address=0x00010cec, savings=2
cm.jalt:
	index=64, sym name=_free_r, address=0x000122c6, savings=30
	index=65, sym name=_malloc_r, address=0x00012bd0, savings=18
	index=66, sym name=__sprint_r.part.0, address=0x00010a1a, savings=18
	index=67, sym name=memset, address=0x00010972, savings=14
	index=68, sym name=strcmp, address=0x00013d90, savings=12
	index=69, sym name=_fflush_r, address=0x00011ce0, savings=10
	index=70, sym name=_sbrk_r, address=0x000138b4, savings=10
	index=71, sym name=memmove, address=0x000133da, savings=8
	index=72, sym name=__malloc_lock, address=0x000134ac, savings=8
	index=73, sym name=__malloc_unlock, address=0x000134ae, savings=6
	index=74, sym name=__sinit, address=0x0001202e, savings=6
	index=75, sym name=__swsetup_r, address=0x00011994, savings=6
	index=76, sym name=strlen, address=0x00013eae, savings=4
	index=77, sym name=_exit, address=0x000145ce, savings=4
	index=78, sym name=memchr, address=0x00013204, savings=4
	index=79, sym name=_localeconv_r, address=0x00012946, savings=4
	index=80, sym name=_fstat_r, address=0x000142e0, savings=4
	index=81, sym name=__sfp_lock_release, address=0x0001203e, savings=2
	index=82, sym name=__locale_mb_cur_max, address=0x000129ba, savings=2
	index=83, sym name=memcpy, address=0x00013294, savings=2
	index=84, sym name=_sbrk, address=0x000146dc, savings=2
	index=85, sym name=_fputwc_r, address=0x0001215a, savings=2
	index=86, sym name=raise, address=0x00013a94, savings=2
	index=87, sym name=atexit, address=0x000141b6, savings=2
	index=88, sym name=__call_exitprocs, address=0x00011a98, savings=2
	index=89, sym name=__sflush_r, address=0x00011b5a, savings=2
	index=90, sym name=__sfvwrite_r, address=0x0001250c, savings=2
	index=91, sym name=_vfiprintf_r, address=0x00010ace, savings=2
	index=92, sym name=_kill, address=0x00014664, savings=2
	index=93, sym name=_realloc_r, address=0x000134b0, savings=2
	index=94, sym name=__swbuf_r, address=0x00013f8e, savings=2
	index=95, sym name=_wcrtomb_r, address=0x000140a6, savings=2
	index=96, sym name=__smakebuf_r, address=0x00012a24, savings=2
	index=97, sym name=strncpy, address=0x00013f12, savings=2
	index=98, sym name=__sfp_lock_acquire, address=0x0001203c, savings=2
	index=99, sym name=abort, address=0x00011a80, savings=2
	index=100, sym name=_isatty_r, address=0x00014322, savings=2

=> the optimal result is to only apply cm.jt to the symbols .L45 and _fwalk_reent.
cm.jt:
	index=0, sym name=.L45, address=0x00010696, savings=10
	index=1, sym name=_fwalk_reent, address=0x0001218c, savings=6

However, if we could change the order of cm.jatl entries, then the optimal result becomes
cm.jalt:
	index=0, sym name=_free_r, address=0x00011f3c, savings=30
	index=1, sym name=_malloc_r, address=0x0001277c, savings=18
	index=2, sym name=__sprint_r.part.0, address=0x0001085c, savings=18
	index=3, sym name=memset, address=0x000107b4, savings=14
	index=4, sym name=strcmp, address=0x000137f8, savings=12
	index=5, sym name=_fflush_r, address=0x00011a46, savings=10
	index=6, sym name=_sbrk_r, address=0x0001339a, savings=10
	index=7, sym name=memmove, address=0x00012f10, savings=8
	index=8, sym name=__malloc_lock, address=0x00012fe2, savings=8
	index=9, sym name=__malloc_unlock, address=0x00012fe4, savings=6
	index=10, sym name=__swsetup_r, address=0x00011720, savings=6
	index=11, sym name=__sinit, address=0x00011d1a, savings=6
```

jnk0le, 2022-11-23:
Note that actual benchmark codes in embench are massively inlined, so there could be false positives when measuring for call compression. (hence mostly stuff like malloc/printf/strcmp etc. is listed here)

jnk0le, 2022-11-23:
Also, previous discussions on partitioning of jalt entries:
here: https://github.com/riscv/riscv-code-size-reduction/issues/173
and here: https://lists.riscv.org/g/tech-code-size/topic/possible_update_to_zcmt/93127613?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,93127613,previd%3D1669218229631957943,nextid%3D1655987884639496880&previd=1669218229631957943&nextid=1655987884639496880

abukharmeh, 2022-11-23:
Can you remind me why we cannot grow the table from the centre ?

linsinan1995, 2022-11-23:
> Can you remind me why we cannot grow the table from the centre ?

Did you mean the idea in this [discussion](https://github.com/openhwgroup/corev-binutils-gdb/issues/44#issuecomment-1248303508)? The entries can grow from the split point, like

```
    ... entries for cm.jt
index 0,  sym=x
             <--- jvt.base
index 64, sym=y
    ... entries for cm.jalt
```

I think it does not fit into the spec which have set a layout for entries,
```
switch(XLEN) {
  32: table_address[XLEN-1:0] = JVT.base + (index<<2);
  64: table_address[XLEN-1:0] = JVT.base + (index<<3);
}
```

linsinan1995, 2022-11-23:
Another similar idea is to allow jvt.base to not point to the table, and then we could prune the cm.jt part like

```
0x0800                                <--- jvt.base
                 ...... (arbitrary instructions)
0x900     index 64, sym=y
                 ......  entries for cm.jalt
``` 
But the base might be negative in this case, which seems not very appropriate(?)

abukharmeh, 2022-11-23:
Yes, I was not convinced then that we cannot do it per the current spec, you can set the tbljalvec at some imaginary point to make the pointers valid. Then you start allocating from 31 down to 0 for jt and 32 up to 254 for jalt ? This make the ideal cacheable area contiguous if we allocate the hottest entries nearest to tbljalvec ?

tariqkurd-repo, 2022-12-07:
Hi everyone, we're talking about 32*4 bytes on a typical system (i.e. RV32).  I'd expect real software on a product to be able to make use of those 32*4 bytes, so I think this is a theoretical argument. I'd also expect software on a real platform to be at least (for example) 100KBytes, so saving ~100 isn't a lot - as by implementing Zcmt we'd hopefully save maybe 5% which is 5KBytes, and more savings from Zcmp, Zcb etc.

So overall I think this is a nice idea, but not very useful in practice.

I'll close it, feel free to re-open it if there's more discussion is needed.

closed 2022-12-07 by tariqkurd-repo