riscvarchive/riscv-code-size-reduction #219: [Question] 64-bit toolchain for Zc*

Abdulwadoodd, 2023-05-10:
I am trying to implement the Zc* extension support in an Open source RISC-V core that generally supports RV64*. Consequently, architecture tests for Zc* (Zcb, Zcmt, Zcmp) needs to be generated/written, ideally for both RV32 and RV64.

[This](https://github.com/riscv/riscv-code-size-reduction/tree/main/toolchain) directory in this repository provides links to download/build a toolchain that supports Zc*. But this toolchain is for only RV32 architecture (and it is working) but I'm unable to compile the Zcb/Zcmt/Zcmp instructions for RV64 architecture using this toolchain. 

Can you please point me to a toolchain (either Clang or Gcc) that supports Zc* extension for RV64 architecture?

I would appreciate any pointers. 
Thanks

tovine, 2023-05-11:
Hi, have you tried the prebuilt toolchain linked in the readme? https://www.embecosm.com/resources/tool-chain-downloads/#corev

It should support both 32 and 64 bit unless I'm mistaken?

> Configured with: ../../gcc/configure --target=riscv32-corev-elf --prefix=/build/workspace/corev-gcc-centos7/install --with-sysroot=/build/workspace/corev-gcc-centos7/install/riscv32-corev-elf --with-native-system-header-dir=/include --with-newlib --disable-shared --enable-languages=c,c++ --enable-tls --disable-werror --disable-libmudflap --disable-libssp --disable-quadmath --disable-libgomp --disable-nls --enable-multilib --with-multilib-generator='rv32e-ilp32e--c rv32ea-ilp32e--m rv32em-ilp32e--c rv32eac-ilp32e-- rv32emac-ilp32e-- rv32i-ilp32--c rv32ia-ilp32--m rv32im-ilp32--c rv32if-ilp32f-rv32ifd-c rv32iaf-ilp32f-rv32imaf,rv32iafc-d rv32imf-ilp32f-rv32imfd-c rv32iac-ilp32-- rv32imac-ilp32-- rv32imafc-ilp32f-rv32imafdc- rv32ifd-ilp32d--c rv32imfd-ilp32d--c rv32iafd-ilp32d-rv32imafd,rv32iafdc- rv32imafdc-ilp32d-- rv64i-lp64--c rv64ia-lp64--m rv64im-lp64--c rv64if-lp64f-rv64ifd-c rv64iaf-lp64f-rv64imaf,rv64iafc-d rv64imf-lp64f-rv64imfd-c rv64iac-lp64-- rv64imac-lp64-- rv64imafc-lp64f-rv64imafdc- rv64ifd-lp64d--m,c rv64iafd-lp64d-rv64imafd,rv64iafdc- rv64imafdc-lp64d--' --with-arch=rv32imac --with-abi=ilp32 --with-bugurl=''\''https://www.embecosm.com'\''' --with-pkgversion=''\''corev-openhw-gcc-centos7-20221031'\'''

Abdulwadoodd, 2023-05-11:
> It should support both 32 and 64 bit unless I'm mistaken.

Apparently, it is not working for 64-bit architecture.

First of all the prefix of the prebuilt toolchain is `riscv32-corev-elf-`, which (probably) means that it only compiles for 32-bit architecture. 
I am compiling my assembly program having `zcb` instructions using the following command (assembly program and linker provided at the end):
```
riscv32-corev-elf-gcc -march=rv32i_zcb -nostdlib -nostartfiles -T link.ld Zc_test.S -o zcb.elf
```
It perfectly compiles the program for **32-bit architecture** and generates the elf which contains the corresponding instructions. Running `riscv32-corev-elf-objdump -d zcb.elf` tells the file format as `elf32-littleriscv`. And this is apparent that it is compiled for 32-bit architecture.

However, when I try to put `-march=rv64i_zcb` in the command above, it throws the following error.

```
cc1: error: ABI requires '-march=rv32'
```

If I'm not wrong then the error message above tells me that the toolchain is unable to compile code and generate elf for 64-bit architecture.

Linker script (link.ld) used:
```
OUTPUT_ARCH( "riscv" )
ENTRY(_start)

SECTIONS
{
  . = 0x80000000;
  .text.init : { *(.text.init) }
  . = ALIGN(0x0100);
  .text : { *(.text) }
  . = ALIGN(0x0100);
  .data : { *(.data) }
  .bss : { *(.bss) }
}
```

Assembly program (Zc_test.S):

```

.text              
        
.global _start
_start:
	nop
	nop
	nop
	lbu x8, 0(x9)
	lhu x9, 0(x10)
	lh  x10, 0(x11)
	sb  x11, 0(x12)
	sh  x12, 0(x13)
	nop
	nop
	nop
	c.lbu x8, 0(x9)
	c.lhu x9, 0(x10)
	c.lh  x10, 0(x11)
	c.sb  x11, 0(x12)
	c.sh  x12, 0(x13)
	nop
	nop
	nop

write_tohost:
	li x1, 1
	sw x1, tohost, t5
	j write_tohost

.data 
.align 14
	.word 2

.align 12
.section ".tohost","aw",@progbits;                            
.align 4; .global tohost; tohost: .dword 0;                     
.align 4; .global fromhost; fromhost: .dword 0; 

```

Please give me pointers to (either) build the 64-bit toolchain that can compile Zc instructions. 

Generally, in [riscv gnu toolchain](https://github.com/riscv-collab/riscv-gnu-toolchain) we can compile for both 64-bit and 32-bit architecture using 64-bit toolchain but we can only compile for 32-bit architecture using 32-bit toolchain. But currently it lacks Zc extension support.

simonpcook, 2023-05-11:
Hi @Abdulwadoodd 

The error message you are seeing is because your command line is missing the `-mabi` option to set to a 64-bit ABI, and the tools default to a 32-bit ABI.

If I take your command line and put `-mabi=lp64` on the end, I can correctly build your file, example output below:

```
simon@hartnell$ ./corev-openhw-gcc-ubuntu2204-20230504/bin/riscv32-corev-elf-gcc -march=rv64i_zcb -nostdlib -nostartfiles -T link.ld Zc_test.S -o zcb.elf -mabi=lp64
```

and disassembly:
```
simon@hartnell$ ./corev-openhw-gcc-ubuntu2204-20230504/bin/riscv32-corev-elf-objdump -d -M no-aliases zcb.elf

zcb.elf:     file format elf64-littleriscv

Disassembly of section .text:

0000000080000000 <_start>:
    80000000:	0001                	c.addi	zero,0
    80000002:	0001                	c.addi	zero,0
    80000004:	0001                	c.addi	zero,0
    80000006:	8080                	c.lbu	s0,0(s1)
    80000008:	8504                	c.lhu	s1,0(a0)
    8000000a:	85c8                	c.lh	a0,0(a1)
    8000000c:	8a0c                	c.sb	a1,0(a2)
    8000000e:	8e90                	c.sh	a2,0(a3)
    80000010:	0001                	c.addi	zero,0
    80000012:	0001                	c.addi	zero,0
    80000014:	0001                	c.addi	zero,0
    80000016:	8080                	c.lbu	s0,0(s1)
    80000018:	8504                	c.lhu	s1,0(a0)
    8000001a:	85c8                	c.lh	a0,0(a1)
    8000001c:	8a0c                	c.sb	a1,0(a2)
    8000001e:	8e90                	c.sh	a2,0(a3)
    80000020:	0001                	c.addi	zero,0
    80000022:	0001                	c.addi	zero,0
    80000024:	0001                	c.addi	zero,0

0000000080000026 <write_tohost>:
    80000026:	4085                	c.li	ra,1
    80000028:	00005f17          	auipc	t5,0x5
    8000002c:	fc1f2c23          	sw	ra,-40(t5) # 80005000 <tohost>
    80000030:	bfdd                	c.j	80000026 <write_tohost>
```

closed 2023-05-11 by Abdulwadoodd