riscvarchive/riscv-code-size-reduction #221: Behavior of -march=zcf on rv64. Behavior of F+Zce on RV64.

topperc, 2023-06-26:
Since Zcf doesn't contain any instructions on RV64. Should toolchains reject it or accept it?

The spec says "Specifying Zce with F includes Zca, Zcb, Zcmp, Zcmt and Zcf". Zcf doesn't contain any instructions on rv64. Should toolchains only enable Zca, Zcb, Zcmp, Zcmt for Zce on rv64?

tariqkurd-repo, 2023-06-26:
is there any precedence for empty extensions @aswaterman ? or is this the first time?

aswaterman, 2023-06-26:
I don’t believe there is such a thing, even though it would be harmless. For my part, I’d favor rejecting it in RV64. Whichever is chosen should be clearly reflected in the spec.

tariqkurd-repo, 2023-06-28:
yes I think that makes sense, although the wavefront spec isn't in great shape as it has incomplete SAIL code in it.
@Alasdair can we remove the SAIL code so that I can update the spec and re-release it?

topperc, 2023-06-30:
This my LLVM patch. https://reviews.llvm.org/D153742 It includes an error for Zcf on RV64. If Zce is specified with F, it only enables Zcf for RV32.

Couple additional questions.
Does Zcf imply F and does Zcd imply D?

topperc, 2023-07-10:
Any progress on getting the spec updated?

tariqkurd-repo, 2023-07-11:
https://github.com/riscv/riscv-code-size-reduction/pull/223

I gave up on getting the SAIL fixed so I reverted that PR, so I had a working head of main again.
Pleae have a look at the PR.

pz9115, 2023-07-11:
> This my LLVM patch. https://reviews.llvm.org/D153742 It includes an error for Zcf on RV64. If Zce is specified with F, it only enables Zcf for RV32.
> 
> Couple additional questions. Does Zcf imply F and does Zcd imply D?

As fld/fsd(sp) instruction description in ISA-spec, you have to use f/d with zcf/d.
But not sure if we need to add the imply relations with them, I just add an error to remind people missing f/d when using zcf/d.

pz9115, 2023-07-13:
> This my LLVM patch. https://reviews.llvm.org/D153742 It includes an error for Zcf on RV64. If Zce is specified with F, it only enables Zcf for RV32.
> 
> Couple additional questions. Does Zcf imply F and does Zcd imply D?

Adding dependencies can make it easier for users to directly use zcf/d extensions，
So I added the depending rules in this PR https://github.com/riscv/riscv-code-size-reduction/pull/224