riscvarchive/riscv-code-size-reduction #226: Zc extension is enabled, the behavior of the .option

bigmagic123, 2023-09-19:
I want to restrict the generation of Zc in a segment of assembly code using the .option.

When I write code like this：

```
.option push
.option arch, -zca
.option arch, -zcb
    add x3,x3,x27
.option pop
```

Passing the argument "arch=rv32ima_zca_zcb" during compilation.

zca remains enabled.

When I browse through the code, I notice that:

https://github.com/bminor/binutils-gdb/blob/master/bfd/elfxx-riscv.c

```
  {"zcf", "zca",	check_implicit_always},
  {"zcd", "zca",	check_implicit_always},
  {"zcb", "zca",	check_implicit_always},
```

When I enable zcb, zca will be included by default.

why it was designed this way?What would be the appropriate usage?

jnk0le, 2023-09-19:
Zca is basically C extension without floating point load/stores

KYG-yaya573142, 2023-10-03:
Based on the spec, all Zc* depend on Zca.
But aren't Zcb & Zca re-enabled by the `.option pop`?

tovine, 2023-10-24:
> I want to restrict the generation of Zc in a segment of assembly code using the .option.
> 
> When I write code like this：
> 
> ```
> .option push
> .option arch, -zca
> .option arch, -zcb
>     add x3,x3,x27
> .option pop
> ```
> 
> Passing the argument "arch=rv32ima_zca_zcb" during compilation.
> 
> zca remains enabled.

Maybe the issue is that you're trying to disable `zca` before disabling `zcb` so the tools don't allow you to do this because `zca` is a dependency of `zcb` (which is still enabled at that point)?
Does it work if you reverse the order and first disable `zcb` and then `zca`?