riscvarchive/riscv-code-size-reduction #228: c.ebreak still requires C?

timsifive, 2023-10-19:
`c.ebreak` is a 16-bit breakpoint instruction, which is required to set breakpoints on 16-bit instructions. However, it is only defined as part of the C extension. That means that if you implement one of the Zc extensions defined here but not all of C, that you cannot set breakpoints in that code.

Did I miss something, or is this an oversight?

jnk0le, 2023-10-19:
Zca should include it

>Zca
Zca is all of the existing C extension, excluding all 16-bit floating point loads and stores: c.flw, c.flwsp,
c.fsw, c.fswsp, c.fld, c.fldsp, c.fsd, c.fsdsp.

pdonahue-ventana, 2023-10-19:
In addition to c.ebreak being in Zca, if you implement Zce, Zcf, Zcd, Zcb, Zcmp, and/or Zcmt then you must also implement Zca (since they're all listed as depending on Zca).  So it all works out.

timsifive, 2023-10-19:
I just also noticed what @pdonahue-ventana commented. All the Zc* require Zca, so you always have `c.ebreak`.

As far as the debugger is concerned, that means:
1. If any 16-bit instructions are implemented, then c.ebreak is also implemented.
2. It is possible to have c.ebreak implemented without misa.C being set.

Is that right?

pdonahue-ventana, 2023-10-19:
That sounds right to me.  Item 2 is a new feature (ignoring the fact that misa is allowed to not be implemented so theoretically you could have always implemented C but had the whole misa register hardwired to 0).

tariqkurd-repo, 2023-10-20:
yes it's certainly true that if you have any 16-bit encodings (and follow the rules) you get c.ebreak

tariqkurd-repo, 2023-10-20:
However I'm not really clear on the rules for setting misa.C. @aswaterman ? Is specifying Zca enough set set misa.C? Do you have to also include everything that C would include in this config - so Zfa, Zfd too to quality if you specify F, D?

So would Zca with no F / D be enough to set misa.C?

I think we need some rules and I can add them into the Zc spec if need be

aswaterman, 2023-10-20:
The C bit corresponds to the C extension, so the rule follows from that: the C bit being set means the C extension is present. (Since Zca == C extension when F/D aren’t present, it would be correct to set misa.C in that case, for example.)