riscvarchive/riscv-code-size-reduction #231: Integration of Zc into the Unprivileged spec.

wmat, 2024-01-31:
This is a heads up that I've began to integrate the Zc* spec into the Unprvileged spec.  It's currently on a Branch of riscv-isa-manual here:  https://github.com/riscv/riscv-isa-manual/tree/zc

After I clean up the asciidoc a bit, I'll be issuing a PR against the riscv-isa-manual tree.  Once it's merged, the specification in this repo will need to be made read only and information added to point future contributors and consumers of Zc* at the Unprivileged spec.

tariqkurd-repo, 2024-01-31:
Excellent news, thanks

wmat, 2024-02-06:
Where are the 32-bit equivalent instructions for the Zcb instructions?  I see references in the spec such as [insns-lbu] in the c.lbu section that are broken links.

I'm going to issue a PR on the riscv-isa-manual repository to pull in the Chapter today.

wmat, 2024-02-06:
Please see https://github.com/riscv/riscv-isa-manual/pull/1216

tariqkurd-repo, 2024-02-06:
Those broken links are intended to point to the 32-bit descriptions of the instructions, and were a best guess of what the links would be

wmat, 2024-02-20:
I'd like to request review of the Zc spec as an integrated chapter within the riscv-isa-manual.  You can view the Unprivileged Spec as a PDF [here](https://github.com/riscv/riscv-isa-manual/actions/runs/7802694882/artifacts/1259678476). @tariqkurd-repo

tariqkurd-repo, 2024-02-22:
Hi @wmat I put several comments on the PR

wmat, 2024-02-22:
Thanks @tariqkurd-repo , I'm actually refactoring the integration as we would like the content to be one file rather than several small files to be consistent with all other chapters.

tariqkurd-repo, 2024-02-22:
Ok. Let me know when that's done and I'll have another look 👍

wmat, 2024-02-22:
OK, all done.  You can grab the PDF [here](https://github.com/riscv/riscv-isa-manual/actions/runs/8007230535/artifacts/1266957313).  Let me know what to do with all of the broken refs throughout that I believe you'd mentioned were placeholders.  Also, I renamed the chapter.  Should but didn't add the Version to it as I wanted to confirm that the current version is 1.0.4-2 with you.

wmat, 2024-02-29:
As of Feb. 29, 2024 Zc* is now integrated into riscv-isa-manual.

closed 2024-02-29 by wmat